chundermike/rpi-fruitbox #1: error while loading shared libraries

bukomail, 2017-12-17:
Hi @chundermike,
first of all thanks for sharing this project.
I just wanted to let you know that I've failed the first two times installing fruitbox.
I followed the steps and both times I got the “error while loading shared libraries”, it was different libraries, sorry I didn't copy the whole message. But it didn't allow me to open fruitbox, the help or even the documentation.
Then I tried one more time installing the libraries separately and it did work. So, maybe was me doing something wrong (even the steps are clear).
Anyway, now is working properly on a Raspberry Pi 1 Model B+

I also have a skin that I would like to share with you to take a look.

thanks

chundermike, 2017-12-17:
Hi,

Sorry to hear that you had problems with libraries...hopefully I haven't missed something out in the installation instructions but glad that you eventually got it working.  I did run through the installation myself on a fresh raspian install and it worked ok, but if anyone else reports library problems I will make sure I take a close look at it.

Very interested to see your skin...would you be happy for me to add it the the skins directory so that me and others could use it?  If so, I'm not sure what the best way of getting the files to me would be since I am rather new to this!

Many thanks for your interest in fruitbox...it gives me real pleasure to know that someone else is enjoying it, and knowing that I can give something back to the open software community :)

All the best,
Mike

bukomail, 2017-12-17:
Hi Mike,
I don’t know if I can reply here

check check

Álvaro

chundermike, 2017-12-17:
Hi

You could always zip the skin files and send them to me at michael (underscore) kingsley (at) hotmail (dot) co (dot) uk...

Mike

bukomail, 2017-12-18:
Ok, still work in progress, but I will finish it soon

closed 2017-12-18 by bukomail