chundermike/rpi-fruitbox #3: Fruitbox not seeing .mp3 files

dirtylarry10, 2018-01-28:
Ive tried everything I can think of (noob here)...I changed both the .cfg files to show the folder I have my .mp3 files in ./Music/DL, but when I run fruitbox to start it there is an error that says there are no .mp3 files in the directory. Any suggestions?

Thank you!!

dirtylarry10, 2018-01-28:
here are the commands:
pi@raspberrypi:~ $ cd rpi-fruitbox-master
pi@raspberrypi:~/rpi-fruitbox-master $ chmod +x ./fruitbox
pi@raspberrypi:~/rpi-fruitbox-master $ ./fruitbox --cfg skins/WallJuke/fruitbox.cfg
here is the error:
nitialising audio ...
Loading database '../fruitbox.db'...0 songs found
ERROR: database file '../fruitbox.db' contains no songs.  Please delete it and re-run fruitbox, making sure the MusicPath parameter in your configuration file points to a directory which contains some MP3 files.

My Music file path in the CFG file is set at:
/home/pi/Music/DL in both the cfg files mentioned in the instructions.

it flickers for a second like its trying to launch but then comes right back to the error listed above.

Larry

JLanza89, 2018-01-28:
hello, you had the same problem

I have the music in the MUSIC folder.

all you have to do is delete the file "fruitbox.db" and run the command "sudo ./fruitbox --cfg skins / WallJuke / fruitbox.cfg" again

JLanza89, 2018-01-28:
I'm new to this and I'm using the translator to write, I hope I can help

dirtylarry10, 2018-01-29:
JLanza89, thank you, that did the trick...I was looking for the .db file in the wrong area. In the instructions READ ME it states to delete this (If you have one) and since I was looking in the wrong spot and (obviously) didnt see it, I was confused.

Thank You

chundermike, 2018-02-26:
Glad all sorted now

closed 2018-02-26 by chundermike