chundermike/rpi-fruitbox #7: "Now playing" text in Modern skin

gavinsj, 2018-02-17:
Hi, I'm fairly new to all this so forgive me if i have missed something obvious, but i'm using the Modern skin in fruitbox and there is some large, permanent, "Now playing" text in the central, lower status window (left hand side of the window with the spinning record in).  I can't find the source of this white text anywhere, but i'd like to remove it if i can.  I have looked in status0.txt, status1.txt and fruitbox.cfg but i cant find any sign of it.  It isnt part of the foreground image either.  Does anyone know the source or how i can remove it?

chundermike, 2018-02-26:
Strange - I can't see that myself!  Have you got the latest Modern skin files from GitHub?  Is it possible to send a screenshot of it?

Thanks,
Mike

chundermike, 2018-02-26:
Please update your Modern skin and fruitbox to the latest versions in Githib...fruitbox versions 1.3 and earlier had some status $variables named $IF_SONG_NOT_PLAYING etc and these were renamed $IF_NOT_SONG_PLAYING etc for consistency with the other variables. If you are using fruitbox v1.4 or above with an old Modern skin then these old $variable names won't be recognised and so will be printed as normal text, in which case you may see the text "NOT PLAYING" on screen.

chundermike, 2018-03-17:
Modern skin status0.txt now fixed

closed 2018-03-17 by chundermike

rooktheklown, 2018-06-21:
This is also occurring with record_spin.txt - It has "$IF_SONG_NOT_PLAYING", resulting in "NOT_PLAYING" showing in that window.

chundermike, 2018-06-21:
Now fixed.  Thanks for the spot.

rooktheklown, 2018-06-21:
No problem. Glad to help.