chundermike/rpi-fruitbox #9: "Z" being replaced with "A" on song titles - Modern skin

rooktheklown, 2018-06-21:
I've noticed that when using the Modern skin, the letter "Z" is replaced with the letter "A" in the song titles. This results in titles such as "CRAAY TRAIN", "CRAAY FOR YOU",  and "CRAAY LITTLE THING CALLED LOVE". 

The issue is with the font itself, as the uppercase "Z" is a duplicate uppercase "A". The easiest fix seems to be changing the capitalize parameter to "false", since the lowercase letters are functionally identical to the uppercase, except that there is an actual "Z". 

I'll test this when I get home later to confirm.

chundermike, 2018-06-21:
Thanks for spotting that...whatever fix you have, please let me know and I'll update the repo.

Cheers,
Mike

rooktheklown, 2018-06-21:
I just tested, and changing the capitalize parameter to "false" fixed the issue, since the lowercase letters are identical to the uppercase (aside from Z, of course).

That said, it is pretty funny to have songs showing up as "CRAAY". I may put it back when I want to mess with people.

PrdAmrican, 2019-01-04:
Wish I'd seen these posts before I pulled all my hair out, LOL.  I think I have fixed the font...
[TravellingFixed.zip](https://github.com/chundermike/rpi-fruitbox/files/2727695/TravellingFixed.zip)

BTW, love Fruitbox.

chundermike, 2019-01-04:
Brilliant thanks.  Modern skin now updated (changed font reference in fruitbox.cfg) and uploaded TravellingFixed.ttf and deleted Travelling.ttf

closed 2019-01-04 by chundermike