chundermike/rpi-fruitbox #10: How can I uninstall fruitbox

ebaystudio, 2018-07-11:
Im trying to clean up my menu. How can I uninstall fruitbox. Thanks.

ebaystudio, 2018-07-12:
Can it be uninstalled?

rooktheklown, 2018-07-12:
Do you have it installed as part of a retropie installation (i.e. Motion Blue v6)?

ebaystudio, 2018-07-12:
Yes it came on a Motion Blue clone. This is the img. https://www.arcadepunks.com/128gb-retropie-4-4-super-display-epic-build-pi-3-b-from-mrharias/

ebaystudio, 2018-07-14:
Do you need more info?

rooktheklown, 2018-07-14:
You probably need to reach out to whomever made the image that you installed, since they included this as part of the image. Even if there is a way to manually delete the software itself, the menu item might still be there.

ebaystudio, 2018-07-14:
I tired that. No reply. Thanks anyway for your time. Ill goof with it until I can figure it out.

rooktheklown, 2018-07-17:
So the image you used was based on this one:
https://www.arcadepunks.com/motion-blue-version-6-stretch-b-version-from-david-marti/

I would reach out to the person that created this image (David Marti), since he is who first included fruitbox with Motion Blue. If anyone knows how to remove it, it'd be him.

closed 2019-02-06 by chundermike