chundermike/rpi-fruitbox #19: Update source code?

6alileo, 2019-02-01:
Hello - any chance you can update your git with the source code so we can build it for the latest? This is for the Odroid community which is running a fork of RetroPie called "RetroArena" specific for Odroid XU4

chundermike, 2019-02-01:
Yes sorry...will update the source code over the weekend

chundermike, 2019-02-01:
By the way, did you solve the sluggish behaviour (lack of GPU rendering)?

6alileo, 2019-02-01:
It's still sluggish in regards to input. Graphics wise, it's not too bad but could be smoother. I think once the input is resolved, it'll feel better. I'll have to see whats that like.

6alileo, 2019-02-01:
But I did figure out the way to configure inputs, that was my issue initially. But the new way seems better.

chundermike, 2019-02-03:
Source code for v.1.4 is now updated

closed 2019-02-03 by chundermike