chundermike/rpi-fruitbox #22: starts and go's back to the menu

jasclo, 2019-02-05:
i can not get this to work it starts and then go's back to the menu

chundermike, 2019-02-06:
Hi
Can you be more specific? When it starts how far does it get? I.e. does it show the jukebox screen or just the spinning logo? Also what menu does it go back to?

Thanks
Mike

jasclo, 2019-02-06:
Just the spinning logo

chundermike, 2019-02-06:
Perhaps it can't find any songs... what error message did it print?

jasclo, 2019-02-06:
Can't tell it goes so fast back to emulation station I am using arcade only
build

chundermike, 2019-02-06:
Hi,
It's almost certainly not finding any songs.  You need to check that the MusicPath parameter in the skin config file is pointing to a directory where your MP3 files are stored.

jasclo, 2019-02-06:
OK I will try again thanks

closed 2019-02-24 by chundermike