chundermike/rpi-fruitbox #24: Fruitbox Skin change

Ed707, 2019-02-24:
Hello
Is there a way to change skins from WallJuke to Wurly skin. Could anybody help on this? I've looked in the rpi-fruitbox-master-skin directory on my SD Card image and I don't have the Wurly Skin.  Could anybody help me please?
@Eddie Rivera

kramg, 2019-02-26:
The Wurly skin was added in v1.12 so I'm guessing that your version of fruitbox is earlier than v1.12?

If so you might want to just do a complete reinstall using the instructions in the README.md section

If you really just want the Wurly skin, I'm afraid I'm not sure how to grab just a subdirectory from GitHub so I would just download the current repo using:

wget https://codeload.github.com/chundermike/rpi-fruitbox/zip/master

Then unzip the master archive and copy the rpi-fruitbox-master/skins/Wurly directory from the resulting rpi-fruitbox-master directory into your ~/rpi-fruitbox-master/skins/ directory

I hope that helps.

Cheers,

- gjk

Ed707, 2019-02-26:
Hey thanks for your reply, would you have the scripts? I’m a noob but I
know how to type them if you give them to me.  Thanks for your help.
@Eddie

kramg, 2019-02-26:
Just follow the README.md at https://github.com/chundermike/rpi-fruitbox

It will walk you through the whole process.  

- gjk

closed 2020-04-13 by chundermike