chundermike/rpi-fruitbox #27: Feature Request - Designate sound device - Retraction (or deferment)

kramg, 2019-03-12:
Hi Mike,

Back on 20 Feb I requested a feature to "add a parameter to designate the sound device used for the sounds defined in the [sounds] section of fruitbox.cfg".  If you haven't already got it sussed, please, disregard my request or maybe just put it on the back burner for now.

I had made the request thinking that it would be trivial to add the -a switch when you called mpg123 since (I thought) you were using mpg123 for playing the music files and allegro for playing the sound effects.  Based on your response along with a little digging into your code I now realise that is not the case and that my ask was greater than I had imagined.  If you haven't already got it sorted and no one else is calling for this feature, I would like to maybe do a little ground work on this while you work on more pressing issues.

If I understand things correctly you initialise allegro's audio addon in fruitbox.cpp then feed it sound effects in engine.cpp or mp3 audio streams (decoded using libmpg123) in audio.cpp.  Determination/initialization of the actual sound device is handled internally within allegro.  What I would like to do is see if I can figure out if and how you can specify what voice (sound device) allegro chooses then see if I can produce a snippet of code that can reliably play a sound where I want.  If I get that far I will touch base with you on how to integrate it back into fruitbox.

Cheers,

- gustav

chundermike, 2019-03-13:
Hi Gustav

Yes your are correct... fruitbox uses allegro for the both the sound effects and mp3 playback,  and chooses the voice internally, so it's not a trivial fix.  I did have a quick look at changing the code but as you rightly assumed I am working on other enhancements at the moment so didn't give it a lot of time.  If you would like to investigate and feedback any code changes or ideas to me I would really appreciate that.

Good luck!

Many thanks
Mike.

kramg, 2019-03-13:
Thanks!  I'll give it a go.  I'll have to warn you though it may take a while - I work slow and am easily distracted by things like, umm, well, fruitbox!

kramg, 2019-04-05:
test of [refering](https://liballeg.org/a5docs/trunk/config.html) to web pages.

kramg, 2019-04-05:
Test of a 

> quote

not sure about this one

kramg, 2019-04-05:
UPDATE:

After several weeks of chasing down far too many dead-ends I have been unable to find any documentation that addresses using multiple sound cards with Allegro.  In addition, my as yet limited grasp of C++ is hampering my ability to engage in brute force experimentation as I often don't know whether I'm fighting a C++ issue or an Allegro issue when attempting to compile my test programs.  As a result I am very close to throwing in the towel on this issue.

What I have discovered so far:

According to the [Allegro 5 manual](https://liballeg.org/a5docs/trunk/audio.html)

>Voices represent audio devices on the system.

and

>Typically, you will only use one voice and attach a mixer to it.

This would imply that while atypical, it **is** possible to use more than one audio device.  Unfortunately there is nothing in the manual or elsewhere that I could find that gives an example or even discusses that possibility.

The [Allegro 4 manual](https://liballeg.org/stabledocs/en/alleg024.html) discusses detecting the sound driver but does not touch on assigning a voice to a specific driver.  For unix based systems it seems to imply that selection is limited to the [sound system](https://liballeg.org/stabledocs/en/alleg038.html#DIGI_*/Unix) e.g. oss, pulse, alsa, etc. I couldn't find anything in the [Allegro 5](https://liballeg.org/a5docs/trunk/) manual that even went into that much depth.

A possibly outdated entry in the [allegro wiki](https://wiki.allegro.cc/index.php?title=Using_TiMidity%2B%2B_with_ALSA_raw_MIDI) indicates that audio devices can be specified in allegro.cfg which the [allegro 4 manual](https://www.allegro.cc/manual/4/a-general-introduction-to-allegro/configuration/) implies is read automatically.  The manual also indicates that allegro.cfg can be created by hand or by using a setup utility and that details of the config file format can be found at docs/txt/allegro.txt.  I was unable to find that file anywhere in the manual or the wiki, but, am wondering if it might be somewhere in the allegro 4 source code.  More to the point for Fruitbox, the [allegro 5 manual](https://liballeg.org/a5docs/trunk/config.html) indicates that configuration files are still used, but, it is unclear if the file is read automatically or only when requested.  It also does not delve into the format of the sections within the file or even what sections are possible.

The allegro vivace [tutorials](https://github.com/liballeg/allegro_wiki/wiki/Allegro-Vivace) on github have not as yet been updated to include sound.  The sound tutorial on the [old wiki](https://wiki.allegro.cc/index.php?title=Allegro_5_Tutorial/Addons/Audio) covers basic sound initialization, but, does not touch on the use of voices. 

The most discouraging bits came from the allegro forums where I stumbled upon [this post](https://www.allegro.cc/forums/thread/590131/651679#target) by a long time forum member which indicated that
> You don't select the voices number yourself... 

and the last three posts, also from long time forum members, in a thread on creating an ALLEGRO_VOICE object in Allegro5.  Beginning from the [third from the last](https://www.allegro.cc/forums/thread/616714/1028320#target) they indicate that

> There is only one voice in allegro 5

> ALLEGRO_VOICE is only needed in very special situations - in fact could probably say it is a mis-feature we only still have for historic  reasons.

and ask

> How do you choose the sound driver???

What I have done so far:

Using the tutorial referenced above I have compiled a test program which, at this point, can only output sound to the default sound card.  I have failed horribly in my attempts to detect the driver used or query the voice about it's capabilities.  The former I suspect because the get_digi_driver command from allegro 4 no longer exists in allegro 5 and the latter I suspect because of my floundering C++ skills.  I will continue this effort by attempting to see if allegro 5 will automatically pick up the desired sound card from a configuration file and then move on from there.  However far I get by this Sunday, 7 April, I will commit to the github branch I have set up for this development effort.

I have attempted to post a question regarding sound card selection on the allegro programming forum but the registration page is not working.  I am awaiting a response from their support team.

One final thing to note is that even if I get this working with two separate sound cards, I'm not sure it will work for those with only the rPi built in sound devices.  In my initial attempts I have been unable to have both the HDMI and 3.5mm port active at the same time.

- gjk

kramg, 2019-05-19:
I have made no further progress with assigning voices in Allegro so am closing this issue.

Although I haven't tried it, I think you may be able achieve the desired effect by playing the desired  "sound effects" using an external program triggered by the status [output] parameter.

- gjk

closed 2019-05-19 by kramg

chundermike, 2019-05-20:
Thanks for your efforts into investigating this.  Juy to let tot know I have also been

kramg, 2019-05-20:
Happy to help, just wish it had panned out.  I still learned lots so don't consider it a wasted effort.
I had come to a similar conclusion, albeit using an external call to mpg123 vice separate libraries, but, didn't feel comfortable  tackling it myself and  even less comfortable  suggesting you do such a major rewrite  :-)  As my C skills are progressing at a a glacial rate I thought I would attempt a simpler mod first!