chundermike/rpi-fruitbox #28: Error: Couldn't open configuration file 'granite' for input

GRVLK, 2019-06-24:
Hey @chundermike ,
Let me start by saying I love the idea and was totally wowed when I saw the videos of the software in action.  I've been trying to get mine up and running for a few weeks off and on, but I keep getting 'Error: Couldn't open configuration file 'x' for input'.  In this case x = granite, miketv, wurly.  Doesn't matter which skin I choose I get the same error.  I've tried re configuring the buttons and but no luck.  I'm using a Raspberry Pi 3B+ and a logitech K400 wireless keyboard,  I'm kind of a noob when it come to pi so I apologize if I'm missing something obvious.  At this point I'm about to reinstall raspbian and fruitbox and see if I have any more luck with that.  Any help would be appreciated

chundermike, 2019-06-24:
Hi

Sorry to hear you're having trouble.  Could you tell me exactly what command you are entering to run fruitbox,  and which directory you are in. Also before running the fruitbox command,  can you type 'ls -al' (no quotes) from the same directory and see if it lists the skins directly. 

Cheers,
Mike

GRVLK, 2019-06-25:
Hi Mike,
I’m using command ‘sudo ./fruitbox --cfg granite’ in the ‘~/rpi-fruitbox-master’ directory.  The list of rpi-fruitbox-master shows the skins directory, but not any skins directly.  When I run the command I get the spinning logo but immediately after I get kicked out with the error.
Thanks for the help.

chundermike, 2019-06-25:
Hi

The cfg option needs to point to the skins configuration file, so you need to use the command 'sudo ./fruitbox --cfg skins/Granite/fruitbox.cfg'

Cheers 
Mike

GRVLK, 2019-06-26:
That did it!  Thanks for the help!

closed 2019-06-26 by GRVLK

cavani76, 2020-09-16:
Ciao GRVLK ho il tuo stesso problema come hai risolto?

chundermike, 2020-09-16:
Try granite.cfg :)

cavani76, 2020-09-16:
Ok Mike ti farò sapere intanto grazie mille

cavani76, 2020-09-17:
Buongiorno Mike ho provato la skin granite ma purtroppo non va'. Altri consigli? Oppure devo reinstallare di nuovo fruitbox?

chundermike, 2020-09-17:
Hi

Please send me the fruitbox.log file and I will check

Thanks
Mike

cavani76, 2020-09-17:
Mike ho aperto il file fruitbox log ma e vuoto

cavani76, 2020-09-17:
Devo scriverci qualche cosa forse?

cavani76, 2020-09-17:
Ho appena aperto il file fruitbox log ma e vuoto forse devo scrivere qualcosa?

chundermike, 2020-09-17:
Strange! It shouldn't be empty.  Can you tell me exactly what your are typing on the command line to run fruitbox?

cavani76, 2020-09-17:
[image]

cavani76, 2020-09-17:
Dopo aver provato ad inserire la skin granite nel file fruitbox log  mi esce questo

chundermike, 2020-09-17:
What did you type in to start fruitbox? I would expect something like...

./fruitbox --cfg skins/granite/fruitbox.cfg

[16003619890351567843382196126717]<https://user-images.githubusercontent.com/71391367/93503174-ff3d6880-f917-11ea-8489-52bd22397d83.jpg>

cavani76, 2020-09-17:
Proprio questo ho digitato

cavani76, 2020-09-17:
Digito prima CD rpi- fruitbox-master/. E poi  digito sudo ./ --cfg skins/granite/fruitbox.cfg

cavani76, 2020-09-17:
Scusa  sudo ./ Fruitbox -- cfg skin/granite/ fruitbox.cfg

cavani76, 2020-09-17:
[image]

chundermike, 2020-09-17:
That says it is looking for the walljuke skin. Check you have a skins directory in the fruitbox directory

[16003644166277896150218883847699]<https://user-images.githubusercontent.com/71391367/93507222-a670ce80-f91d-11ea-8137-91e859cb48de.jpg>

cavani76, 2020-09-17:
C'è sia la cartella skin che il file skin txt

cavani76, 2020-09-17:
A limite provo a reinstallare fruitbox e poi ti aggiorno perché potrebbe esservi qualche errore nel download forse

chundermike, 2020-09-17:
Check the skin file contains all the skins. If so, you are probably not running fruitbox from its home directory. Open a command terminal, then type...

cd ~/rpi-fruitbox-master
./fruitbox --cfg skins/granite/fruitbox.cfg

Or for the skin chooser,  type...

./fruitbox --choose-cfg skins.txt

chundermike, 2020-09-17:
I would recommend Raspian Lite, not Raspian desktop

chundermike, 2020-09-18:
> Ciao Mike ho reinstallato il raspian però ha questo punto mi esce questa scritta come mai?

Looks like a network problem. Check your internet and try again :)

cavani76, 2020-09-18:
Ok grazie mille

cavani76, 2020-09-18:
Come sempre gentilissimo

cavani76, 2020-09-22:
Buongiorno Mike sai dove posso scaricare una versione precedente di fruitbox?  Attualmente stavo usando la 1.6 ma purtroppo mi da problemi grazie

chundermike, 2020-09-22:
All previous versions are available on GitHub...click on the fruitbox executable file and select history, which will navigate you to all releases.

Out of interest , what problems are you having with the latest release?

cavani76, 2020-09-22:
Sulla versione 1.6 ho mappato i pulsa nti ma non riesco a capire bene dove inserire gli MP3 e come modificare la Path music nella skin desiderata

cavani76, 2020-09-22:
Forse e per questo che non parte fruitbox

cavani76, 2020-09-22:
Il resto dei passaggi li ho fatti perfettamente

chundermike, 2020-09-22:
That hasn't changed in version 1.6.

You can put your mp3 files anywhere, and point to them using the MusicPath parameter in the skin cfg file (or --music-path command line argument).

For example...in a command terminal type...

cd /home/pi
mkdir music

Then copy all your mp3 files and/or mp3 folders to /home/pi/music

Then in the skin cfg...

MusicPath = /home/pi/music

...in bocca al lupo!

cavani76, 2020-09-23:
Scusa Mike se ti disturbo ancora ma credo che avrei bisogno di un video tutorial completo magari così faccio tutti i passaggi corretti e riesco a completare! Tu non potresti inviarmelo così quando completo tutto  ti invio anche la foto del mio jukebox fatto a mano! Grazie mille per il tuo aiuto

cavani76, 2020-09-23:
Con altri lavori nessun problema ma con questo non so non ci rieso

cavani76, 2020-09-23:
Infatti realizzo anche mini cabinet arcade

cavani76, 2020-09-23:
[image]

cavani76, 2020-09-23:
[image]

cavani76, 2020-09-25:
Ciao Mike sono riuscito a fare partire il programma ho caricato anche gli MP3 e funzionano tutte le skins ma ho un problema audio .sto usando delle casse con jack 3.5  inserite nella Raspberry ma niente audio sai come aiutarmi?

cavani76, 2020-09-25:
Forse devo impostare qualche settaggio?

chundermike, 2020-09-25:
Hi,

Yes, you need to modify the system settings. Try ...

sudo raspi-config

And then look for the audio settings. If it's not obvious,  try googling for the solution...it's quite a common problem.

Mike

cavani76, 2020-09-29:
Ciao Mike ho risolto anche il problema audio 👍 volevo chiederti quali MP3 sono ideali  per fare riconoscere l'artista ed il brano

cavani76, 2020-09-29:
Usi qualche formato in particolare?

cavani76, 2020-09-29:
Ultima domanda come faccio a salvare la playlist nel database ho notato che ogni volta che avvio fruitbox deve sempre caricare i brani prima

cavani76, 2020-09-29:
Ciao Mike ho risolto il problema audio👍 volevo chiederti se usi qualche tipo di MP3 in particolare per fare riconoscere a fruitbox gli artisti ed i brani

cavani76, 2020-09-29:
Infine come salvo la playlist nel database

chundermike, 2020-09-29:
Hi

You can save the playlist using the --save-playlist command line argument, and load it back in later using the --load-playlist option. Please read the user guide.

Also, fruitbox will recognise both IDv1 and IDv2 mp3 tags for the song title and artist (and many other tags), so normal mp3 files will work fine, as long as they have the correct metadata in them. You can always add this data yourself to the mp3 files by using free software such as mp3tag (for Windows)

Cheers
Mike

cavani76, 2020-09-29:
👍 grazie mille

cavani76, 2020-11-26:
Ciao Mike come ti avevo promesso ti mando la foto del jukebox da me
realizzato .grazie per il tuo aiuto col software.by Cavani 76

cavani76, 2021-10-16:
> Non riesco a far partire fruitbox… quale versione devo scaricare ???
> Neanche mi finisce di fare l’aggiornamento

Prova la skin walljuke
Versione 6.1

Pegabri, 2021-10-17:
> Ciao Mike ho risolto il problema audio👍 volevo chiederti se usi qualche tipo di MP3 in particolare per fare riconoscere a fruitbox gli artisti ed i brani

Pegabri, 2021-10-17:
Ciao Cavani76, sono riuscito a far partire con WallJuke… ho solo un problema e ti chiedo aiuto. Ho eseguito il comando autostart ed è tutto ok , solo che quando forzo l’uscita alle casse (3.5) … perde il comando di auto start. Se voglio l autostart la musica esce in hdmi na senza auto start esce dalle casse. Come posso fare? 🙏🏻🙏🏻🙏🏻🙏🏻🙏🏻