chundermike/rpi-fruitbox #29: Select same song multiple times

ZS6TVB, 2019-09-03:
It seems that it is possible to select the same song more than once.  From my experience on the old mechanical jukeboxes it would only select a song once.  Is there a way to only put a song in the playlist if it has not been selected for the current playlist previously or only let the user select the song again once it has finished playing, hence not in the current playlist of selected songs any more?

chundermike, 2019-09-03:
Hi
The next release will have a "NoDuplicates" skin parameter which will allow you to achieve this as an option.

closed 2019-09-03 by chundermike

ZS6TVB, 2019-09-03:
Stunning!

Thank you!

chundermike, 2019-09-03:
Just a quick question... on these old jukeboxes, will they use up a credit if you select a duplicate song, or keep the credits the same? I'm just trying to get the behaviour correct.  (If different jukeboxes do different things then I'll have to add another parameter to specify that behaviour too...)

whitebug71, 2019-09-04:
Each played song is a credit. So if you play the same song 3 times, it will cost you 3 credits.

Also a little tip, I don;t know if you know, but when a jukebox is idle for 15-30 min, it normally plays 1 random song. Just to let people know there is a jukebox and attract them to the machine to play music. :-)

chundermike, 2019-09-04:
Thanks.  Assuming that means then if a selection yields a duplicate song which is subsequently not played,  then no credits are taken. 

Fruitbox already has the ability to play a random song after a period of idleness, through the "AutoPlay" and "AutoPlayGap" skin parameters. Check out the user guide for details...

ZS6TVB, 2019-09-04:
You have it all right.

The old mechanical jukeboxes defenitely used a credit for a song selected again.  They had no way to show the upcoming songs in a playlist and the user did not know if the song was already selected or not.  I would therefore suggest the same behaviour.

chundermike, 2019-09-04:
Thanks again. 

In the spirit of fruitbox I'll add another skin parameter to specify whether a credit is taken if a duplicate song is chosen. That way I cover my arse, and it'll allow more flexibility (and in true Brexit fashion allows me to put off making any decisions...)

chundermike, 2019-10-10:
Now implemented in v1.15

ZS6TVB, 2019-10-13:
Great job.

Working like a charm in v 1.15