chundermike/rpi-fruitbox #31: RPI4 Does not seem to work with fruitbox

fokenhelpmy, 2019-09-27:
Hi

Has anyone got this to work with the pi 4.
i keep getting failed to create display errors.

Thanks
Robbie

gaz1971uk, 2019-09-27:
Mine wont go past detecting display,

fokenhelpmy, 2019-09-27:
Got that too after # out some settings in config.txt

gaz1971uk, 2019-09-27:
I had to comment out, was getting an add service error.  It works on my pi3, thinking it may have something to do with having 2 hdmi outputs. But I haven't found anything yet on internet on how to resolve the issue

fokenhelpmy, 2019-09-27:
that the same thing my wife said, had a similar issue with a few other projects.
I really hope someone can help out on this

gaz1971uk, 2019-09-27:
So do I  I bought the 4 just for the extra power for the project , really dont want to have to buy a pi3

Ruben-pottie, 2019-09-28:
Same here, new clean installation.

Ruben-pottie, 2019-09-29:
Update, i might not be hardware related, using the same image (buster) on RPI-3 gives the same error message.

chundermike, 2019-09-30:
Hi

I haven't tried fruitbox on Pi4, but currently migrating to SDL2 from Allegro, so that may help...

chundermike, 2019-10-01:
There may be a solution presented by Karrika on the raspberry Pi forum...

https://lb.raspberrypi.org/forums/viewtopic.php?f=38&t=188723&sid=4d3533a3646a98f35bb939188d4db1fc&start=1050#p1543258

...he seems to suggest that he has compiled fruitbox from the source code for the Pi4, but had to modify the Makefile to overcome some library conflicts. You could try this yourself or post a message on the forum to ask him if he'd be so kind as to issue a pull request to me to add the Pi4 executable to GitHub

Ruben-pottie, 2019-10-02:
I'll try that this evening. However I already compiled from source on rpi4 but did not encounter the said library issues

Ruben-pottie, 2019-10-02:
trying to build with this makefile gives the following: 
pi@raspberrypi:~/rpi-fruitbox-master/build $ make
g++ -Wall -o ../fruitbox fruitbox.o playqueue.o audio.o config.o database.o display.o engine.o pages.o status.o userguide.o bitmap.o joystick.o input.o gpio.o -O2 -std=c++14 -D_RPI -lmpg123 -lm -ldl -lallegro -lallegro_memfile -lallegro_font -lallegro_ttf -lallegro_audio -lallegro_image -lallegro_video -lallegro_primitives -lallegro_acodec -lpthread
/usr/bin/ld: gpio.o: in function `GpioClass::OpenDevice()':
gpio.cpp:(.text+0xa0): undefined reference to `bcm_host_get_peripheral_address'
collect2: error: ld returned 1 exit status
make: *** [Makefile:17: ../fruitbox] Fout 1

content of the makefile:
`CPP         = g++
CFLAGS      = -O2 -std=c++14 -D_RPI
INC         = -I/opt/vc/include
DEPS        = fruitbox.hpp fruitbox_types.hpp pages.hpp status.hpp playqueue.hpp audio.hpp config.hpp config_base.hpp database.hpp display.hpp \
              engine.hpp fruitbox_logo.hpp info_bkgnd.hpp chooser_arrow.hpp info_font.hpp status_events.hpp userguide.hpp bitmap.hpp joystick.hpp \
              display_base.hpp input.hpp gpio.hpp
OBJ         = fruitbox.o playqueue.o audio.o config.o database.o display.o engine.o pages.o status.o userguide.o bitmap.o joystick.o input.o gpio.o
LIBS        = -lmpg123 -lm -ldl -lallegro -lallegro_memfile -lallegro_font -lallegro_ttf -lallegro_audio -lallegro_image -lallegro_video -lallegro_primitives -lallegro_acodec -lpthread
STATIC_LIBS = allegro-static-5 allegro_memfile-static-5 allegro_font-static-5 allegro_ttf-static-5 allegro_primitives-static-5 \
              allegro_image-static-5 allegro_audio-static-5 allegro_acodec-static-5 allegro_video-static-5

%.o: %.cpp $(DEPS)
	$(CPP) -c -o $@ $< $(CFLAGS) $(INC)

# ./fruitbox: $(OBJ)
../fruitbox: $(OBJ)
	$(CPP) -Wall -o $@ $^ $(CFLAGS) $(LIBS)
	#$(CPP) -Wall -o $@ $^ $(CFLAGS) $(LIBS) `pkg-config --libs --static $(STATIC_LIBS)`

.PHONY: clean
clean:
	rm -f *.o
	rm -f ../fruitbox
	touch Makefile
	touch *.?pp`

Ruben-pottie, 2019-10-02:
or maybe i made a mistake in de compile commands? Please feel free to share so i can recompile and share you the compiled files.

Ruben-pottie, 2019-10-02:
Compiled succesfully, however the same error occurs. Meanwhile putting the same Buster version in Rpi3 does work,

chundermike, 2019-10-02:
Thanks for the info.   It looks like the Broadcom driver (bcm_...) has a different API for the Pi4.  I'll investigate.

Cheers
Mike

Ruben-pottie, 2019-10-04:
Hi Mike,

just figured something weird out, seems like if i compile without the "-D_RPI" parameter in "CFLAGS      =" i can start the application and it reacts the same way as described by Karrika.

However i am not familiar with the different parameters, but maybe this helps you forward in finding a resolution for this problem?

chundermike, 2019-10-04:
Thanks for the feedback...The -D_RPI option turns off tbe Raspberry Pi hardware specific GPIO code,  which is what tbe Broadcom library refers to.

Cheers
Mike

jjlakis, 2020-03-11:
I have also successfully built fruitbox on Rpi4 with dynamic libs. However I stuck on:
```
ERROR: Couldn't install audio
```

DFF-fred, 2020-06-06:
Hi! Compiled fine on RPI4 but I'm getting "failed to add service - already iun use?" Running on rasbian. Any idea what I could do now?

chundermike, 2020-06-06:
Hi
The Allegro graphics library doesn't work on Pi4 without X11 (desktop), I'm waiting for the Allegro developers to come up with a solution. 

Mike

DFF-fred, 2020-06-07:
Thank you very much for that very prompt reply. I guess I will be waiting as well then ;-)

darkt0rcr3w, 2020-10-07:
Is there now a solution for Raspberry Pi 4?
Or will there be a new version sometime?

chundermike, 2020-10-08:
Hi

Still waiting for an Allegro 5 library update  to implement a non desktop solution.

I'm intending to try a desktop based version soon (before the end of the year)

darkt0rcr3w, 2020-10-12:
Great! I run your system integrated in Retropy. So I would like to upgrade to Raspberry 4 to get the extra performance for games...

DFF-fred, 2020-10-13:
OT to darkt0rcr3w: do you have a write-up of your integration somewhere? thank you.

wstevens0n, 2020-10-18:
I'm using X11 to launch on my Pi4 build using fkms and 64bit.

tar -xvf mpg123-1.24.0.tar
cd mpg123-1.24.0
chmod +x configure
./configure --with-cpu=arm_fpu --disable-shared
make -j4 && sudo make install

cd ~/rpi-fruitbox-master/build # back to the fruitbox build directory

tar -xvf allegro-5.2.4.0.tar
cd allegro-5.2.4.0
mkdir build && cd build
sudo nano ../cmake/Toolchain-raspberrypi.cmake 

_UPDATE CODE_
```
SET(CMAKE_C_FLAGS "-march=armv8-a+crc+simd -mcpu=cortex-a72 -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -funsafe-math-optimizations")
SET(CMAKE_CXX_FLAGS "-march=armv8-a+crc+simd -mcpu=cortex-a72 -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -funsafe-math-optimizations")
```
cmake .. -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchain-raspberrypi.cmake -DSHARED=off
make -j4 && sudo make install

cd ~/rpi-fruitbox-master/build # back to the fruitbox build directory

sudo nano ./Makefile

_UPDATE MAKEFILE_
```
CPP         = g++
CFLAGS      = -O2 -std=c++14 -Wno-psabi
INC         = -I/opt/vc/include
DEPS        = fruitbox.hpp fruitbox_types.hpp pages.hpp status.hpp playqueue.hpp audio.hpp config.hpp config_base.hpp database.hpp display.hpp \
              engine.hpp fruitbox_logo.hpp info_bkgnd.hpp chooser_arrow.hpp info_font.hpp status_events.hpp userguide.hpp bitmap.hpp joystick.hpp \
              display_base.hpp input.hpp gpio.hpp
OBJ         = fruitbox.o playqueue.o audio.o config.o database.o display.o engine.o pages.o status.o userguide.o bitmap.o joystick.o input.o gpio.o
LIBS        = -lmpg123 -lm -ldl -lallegro -lallegro_memfile -lallegro_font -lallegro_ttf -lallegro_audio -lallegro_image -lallegro_video -lallegro_primitives -lallegro_acodec -lpthread
STATIC_LIBS = allegro-static-5 allegro_memfile-static-5 allegro_font-static-5 allegro_ttf-static-5 allegro_primitives-static-5 \
              allegro_image-static-5 allegro_audio-static-5 allegro_acodec-static-5 allegro_video-static-5

%.o: %.cpp $(DEPS)
	$(CPP) -c -o $@ $< $(CFLAGS) $(INC)

../fruitbox: $(OBJ)
	$(CPP) -Wall -o $@ $^ $(CFLAGS) $(LIBS)

.PHONY: clean
clean:
	rm -f *.o
	rm -f ../fruitbox
	touch Makefile
	touch *.?pp
```

make -j4
cd ..

sudo chmod +x ./fruitbox

sudo nano ./exec_fruitbox

_Create script to launch with X_
```
#!/usr/bin/env bash

/home/pi/rpi-fruitbox/fruitbox --button-map /home/pi/rpi-fruitbox/[yourbuttonfile].btn --cfg /home/pi/rpi-fruitbox/skins/[yourskin]/fruitbox.cfg 
```
sudo chmod +x ./exec_fruitbox
sudo mv ./exec_fruitbox /usr/bin

launch with the following command
startx -fg black -bg black -e 'exec_fruitbox' -- -nocursor

**NOTES:**
**-I had to remove .wav files from fruitbox.cfg those crashed the system
-X11 is limited, I had to use xrandr to rotate the window for my setup (if rotation can be implemented it would be appreciated)**

chundermike, 2020-10-20:
Hi Will

That's great.  I'll give it a try myself. Are you using Buster Lite or desktop? And if the latter, are you booting to desktop first?

And yes, I'll put a rotate option in the next release.

Cheers
Mike

wstevens0n, 2020-10-20:
Hey Mike,

I'm using a custom 64bit kernel built on Buster 4.19.81-v8+.

In theory, as long as the symlinks are correct this should work the same. What I've noticed is most packages require Toolchain updates for Rpi64 64bit otherwise they simply break.

My setup is non-desktop, so I am using X window to open my applications. Hence the need for the bash script.

Once the application launches I use xrandr to do adjustments (if this is required) such as rotations and/or transformations to stretch it to 480x800.

Rotational support would be great or have fruitbox only utilize resolution in fruitbox.cfg.

One other item I forgot to mention is that if the theme has sounds referencing wav files it does crash, but that could just be related to my userspace.

Project photo
https://pasteboard.co/Jwx9MF5.jpg

Cheers,
Will

chundermike, 2020-10-20:
Hi Will

Thanks.  FYI you can use the --no-scale command line option to prevent fruitbox scaling to full screen.

Cheers
Mike

https://pasteboard.co/Jwx9MF5.jpg

Xlblaze, 2021-03-16:
Hi Mike

I am also having this issue with the RPI4, have you released an updated version?

Paul

chundermike, 2021-03-16:
Hi Paul,

Sorry, I’ve been concentrating on v2, but still intend to try and find a Pi4 solution.  I’ve given up trying to guess timescales because I’m hopelessly rubbish at estimating how long it all takes ☹

Cheers,
Mike

sergiohf, 2021-05-13:
Anyone manage to install on the rpi4? I manage to compile allegro without errors with wstevens0n tutorial. But the adapted makefile cant compile, and gives me an error. The original makefile compiles but gives me x11 error when i try to run. I am using a rpi4 2gb with raspian buster lite, with Openbox. I cant wait to see it working on the pi4.