chundermike/rpi-fruitbox #32: wireless keyboard problems

bryans49, 2020-01-01:
Hi Mike
thank you for a great program! i have been using this in a custom home built jukebox in my living room for over a year now, and its been working great!! 
I decided to update to ver 1.15. i went and grabbed your new version. i am sorry to say. it no longer works with my logitech K400r keyboard. or any bluetooth input device. i also tried remapping the buttons. and it appears to have conflicts with keys that are not even mapped or used for the skin.
only some of the skins seem to work with a regular USB keyboard. other ones dont work at all anymore. i read through the issues other people were having, and your suggestion to revert back to v.1.14, but you cant seem to do this either, because when you run the install script from ver. 1.14  it points to where the new code is and downloads 1.15 again. I really loved this program and need to get it up and running again. could you please fix the install so i can re-download ver.1.14.
thank you! the hardware i am running this on is a RPI ver 3b+ and its always worked flawlessly.

    Something that would be great to see implemented in a future version would be an on the fly search function, as i have over 2k songs on my fruitbox. that would really be an awesome feature to have!
thank you,
Bryans49

chundermike, 2020-01-17:
Hi,

Sorry for the delay in getting back to you...didn't see the post.

Sorry you've got problems...I'll see if I can get to the bottom of the issue.

In the meantime, to revert back to version 1.14, you can find v1.14 from a web browser on GitHub by clicking on the fruitbax executable, then History, then <>.  This will take you to that version in the repository. Or you can just follow this link...

https://github.com/chundermike/rpi-fruitbox/tree/c7edc3e4ed06689c5aa315cb27f665eeaa9cf693

..then click on the fruitbox executable, then select "download"

Oh, and I am working on a search feature :)

Cheers,
Mike

bryans49, 2020-01-18:
hi Mike !
thats ok. thank you for such an awsome program! thanks for trying to get
this figured out.  and i am looking forward to your new version with a
search function . i have a good size collection of music, and a search
would be great!!
please let me know when your new verion is ready. untill then i will go get
v1.14 and get it up and running again.