chundermike/rpi-fruitbox #33: Select the skin with a touch screen

iOtero55, 2020-02-11:
How to use the --choose-cfg option with a touch screen without a keyboard?

certifiablegeek, 2020-02-28:
Same issue here, I think.  Official Pi 7 inch screen used and all the bottom buttons, vol+-, arrows, random... etc.. non functional.  on purpose? or broken? you decide?

iOtero55, 2020-02-28:
I use that screen and all the buttons work for me. The only thing that doesn't work for me is the --choose-cfg command line option. Mike is already in it and in the next version he will solve it.
To use the touch screen you have to have the skin and fruitbox.cfg very well configured, it is not easy at first.
If you go through the fruitbox forum, maybe we can help you:
[https://93.93.130.214/forums/viewforum.php?f=38&sid=a576b7b604dd4cafcda0db837a643e06](url)

chundermike, 2020-05-18:
As iOtero has mentioned, there is an issue with the choose-cfg option with touchscreens, which he rightly said I am currently working to fix for the next release (v2.00).  I also appreciate that configuring the touch screen inputs for the touch specific skins is a bit of a multi-step process, and not particularly straightforward.  I intend to make this process easier in the next release, too.

Cheers,
Mike