chundermike/rpi-fruitbox #35: error while loading shared libraries: libvorbisfile.so.3 while starting

CoolCash1, 2020-04-14:
I cant seem to launch fruitbox because of a library error (error while loading shared libraries: libvorbisfile.so.3). it says there is no such file or directory. thanks for any help.

chundermike, 2020-05-18:
You may have a mismatch between the OS release and fruitbox version.  On GitHub there are three different fruitbox versions, for Jessie, Stretch and Buster.  Alternatively you could just grab the SD card image which is now available.

Cheers,
Mike

jb0815, 2021-10-02:
Hello,
I have the same issue. However, the image doesn't work with USB Devices on my 3b.
So I tried installing the alternative way. It all worked fine, but I can't start. I'm having the same error (error while loading shared libraries: libvorbisfile.so.3).
I'm on Buster, latest version. I've tried to start with fruitbox-buster or fruitbox-jessie. But the access to those files is denied.

And now I'm lost... :-D I don't know how to get another version by using this link: https://github.com/chundermike/rpi-fruitbox/raw/master/install.sh

Can you help me? Thank you so much :-)

chundermike, 2021-10-03:
Hi

Did the vorbis library install OK? It should be done by the install script, but you could try it manually with...

sudo apt-get install libvorbis-dev

...does that work?

Cheers
Mike

jb0815, 2021-10-03:
Hi Mike,

thanks for your quick response.

I've found another way last night. I took your image, manually mounted my usb stick with fstab and modified the rc.local with root user.

That works fine so far. :-) I still don't know why your built in USB mount doesn't work in my case. But I will try a brand new device in a few days. Maybe the one I use for testing is just too old.

Cheers,
Jens