chundermike/rpi-fruitbox #39: Fruitbox

flynn14, 2020-06-09:
Can anyone help me i follow all the steps to install fruitbox on raspberry pi 3b but cant seem to get the skin config to work so i can map my usb music to my chosen skin aka Wurly

chundermike, 2020-06-12:
If you use the SD image it will expect the music to be on a USB stick. If you want fruitbox to run the Wurly skin rather than the default skin chooser then you'll need to edit /etc/rc.local and change the fruitbox command (from --choose-cfg skins.txt to --cfg skins/Wurly/fruitbox.cfg)

flynn14, 2020-06-13:
Thanks mike but if i copy my music onto my sd card music it works fine but i want to be able to leave my music on my usb drive so i can add more music later, but when i do that it come up with an error music path the path i am using is ./media/usb/Music and nothing sorry to be a pain

chundermike, 2020-06-13:
If you read the userguide.txt there is a section near the end on how to set up the Pi to mount an SD drive automatically and set the music path correctly.

flynn14, 2020-06-13:
Awesome thanks for that i am new to this that doc is an awesome help. Awesome job you have done great system

closed 2020-06-16 by chundermike