chundermike/rpi-fruitbox #40: VGA monitor

gaz1971uk, 2020-06-24:
Running fruitbox image, was working fine but I changed from a hdmi tv to a vga monitor, starts up fine, but as soon as song starts playing monitor goes off, but comes back on after song is finished. I'm at a loss to why. Any ideas? 

Thank you

gaz1971uk, 2020-06-24:
So I've been playing with all the settings within the configure.txt file and no joy. When it boots I select the skin I want. If I leave it alone the screen stays on. 45 secs after a song starts the screen flashes on then off a few times, then goes off. Once song stops the screen comes back on. Also if I escape into the command line the screen returns. 
Please help my head done in

ZS6TVB, 2020-06-24:
Just something coming to mind on my side…

Does the Pi power supply provide enough Amps?  From the info you provided I presume you may have an HDMI to VGA converter with extra power consumption or something of the sorts.

Maybe it will good to check if your power supply is adequate.  dmesg will usually report a message such as "Under-voltage detected" should there be a power supply issue.

try something like dmesg | grep voltage to see any such messages.

ZS6TVB, 2020-06-24:
syslog should also report “Voltage normalised” messages.

try:

cat /var/log/syslog | grep -i voltage

gaz1971uk, 2020-06-24:
Haven't found any issues for power, it was working on to while I did the build using just hdmi. I have hdmi to vga, USB encoder, USB memory stick, and a fan, plus USB wireless keyboard, I'm new to this so I'm trying to get my head round it. I have no symbol in corner, red light stays on. It's the official power supply. I've tried taken it down to bare bones, with just 1 song on memory card for testing. If I don't select a song the screen stays on until auto play, then off. Comes up with no cable connected.

ZS6TVB, 2020-06-24:
Did you check syslog for any power messages?

gaz1971uk, 2020-06-24:
I've checked syslog, but haven't found any power messages.

gaz1971uk, 2020-06-24:
I'm just going to with it must be power as I can't find anything else a the minute,and it makes sense as the problem only occurs when the pin is using USB memory stick.so the question is how do I resolve this problem?

ZS6TVB, 2020-06-24:
Not sure,

Mike may be able to help much more than I can.

I would go through /var/log/syslog and look for any messages that look suspect and google it.

I have used an HDMI to VGA converter on a Raspberry Pi Arcade machine I built using RetroPie causing me tremendous trouble.  In the end I replaced it with a different make converter and never had any troubles since.  My Jukebox uses an HDMI TV display so I can not comment on issues with Fruitbox but it does sound like either a power or HDMI converter issue.  Maybe try a different HDMI converter if you can get your hands on one?

gaz1971uk, 2020-06-25:
Thanks for help.
I've ordered another hdmi to vga adapter. So I'll try that and if that doesn't work I'm going to see about a powered USB hub. Hopefully one of these will solve the problem. Any recommendations on powered USB hubs, I'm in the uk

gaz1971uk, 2020-06-29:
So I've tried a new vga cable and a powered couverture , still no joy, head is really done in now

closed 2020-07-03 by gaz1971uk