chundermike/rpi-fruitbox #42: No sound when set to automatic run

zarmedia1, 2020-07-07:
Fruitbox audio works fine when manually executing program in terminal.  Lines added to bashrc per user guide to auto run, program boots successfully (Pi set to boot to terminal with auto password fill-in), visually looks fine and plays, but no sound.

•audio output of pi configured to headphone out
•amixer set to output to headphone

Again, everything works as it should when manually started.  When auto-booting, no audio.  I want to auto-boot since this will be a standalone device.

I'm somewhat of a novice, so it's safe to assume it could be something low level/obvious that I'm overlooking.  Thanks for the assistance.

Comet-boy, 2022-05-23:
Did you find a solution? I've got a similar issue!

zarmedia1, 2022-05-23:
I don't think so.  Frustrating when developers won't respond to issues that keep the entire thing from working, but it's a freebie, so I get it I guess.

Comet-boy, 2022-05-23:
I’ve just solved my sound issue by creating a “ asound.conf” file as per this article:

 

Setting the default device - AlsaProject (alsa-project.org) <https://www.alsa-project.org/wiki/Setting_the_default_device> 

 

Have you tried doing this??

 

Bruce

zarmedia1, 2022-05-24:
Thanks Bruce.  My original request for help was about 2 years ago, so this project has been long shelved, but I'm interested in bringing it out again because I have some slot machines I want to convert to jukeboxes.  I will definitely refer back to this when that times comes, thanks a lot for sharing!

GRVLK, 2022-06-06:
Try leaving out the 'sudo' on the 2nd line. Not sure why this works but it did for me.