chundermike/rpi-fruitbox #47: Errors on initial fruitbox run

Halen78, 2020-09-23:
Hello,
Today I installed fruitbox on my raspberry pi 3B, and after what looked like a clean and successful install, I did a test run passing no arguments:
./fruitbox

The output was as follows:
**./fruitbox: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by ./fruitbox)
./fruitbox: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by ./fruitbox)
./fruitbox: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ./fruitbox)
./fruitbox: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by ./fruitbox)**

The instructions state very clearly, "It is recommended to start from a fresh OS install (latest Raspbian Buster release)".  But I am installing and running fruitbox on my RetroPie image, which is a couple years old at this point.  So I am not surprised that I am running into issues.  But I am hoping that someone might have advice to get me past the issues without having to scrap my entire RetroPie image.

By running this command, to confirm which GLIBCXX files I am missing:
"strings /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 | grep GLIBCXX"
I see that I only have files up to and including "GLIBCXX_3.4.20"
Whereas, fruitbox clearly needs the next two in the series, GLIBCXX_3.4.21 and GLIBCXX_3.4.22
Similarly, my CXXABI files only go up to CXXABI_1.3.8 (I need CXXABI_1.3.9),
and my GLIBC_ files go up to GLIBC_2.18 (I need GLIBC_2.28).

If there is a "sudo apt-get install" or a wget that I can run to resolve the issue, I would really appreciate a clue.  All that I tried based on hours of Googling have failed me.

If my only answer is that I need to update my entire operating system (to the latest Raspbian Buster release), a good link for a beginner to perform that upgrade would be equally appreciated.

Halen78, 2020-09-23:
Update: I'm currently running Version 8 (jessie)...

pi@retropie:~/rpi-fruitbox-master $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian

chundermike, 2020-09-23:
Just install a fresh buster release and it should work.  On tbe raspberry pi website you can download the latest buster lite image; and there's instructions kn how to burn if to an SD card.

Halen78, 2020-09-24:
@chundermike 
Would you be willing to leave this issue open for a little while, in case other users have successfully gotten past these issues, and gotten Fruitbox running on older jessie-based images?

chundermike, 2020-09-24:
Sure. FYI you can always build a Jessie version yourself...just follow the instructions in the build.txt file in the directory.  Alternatively if you don't want the latest release,  there's a Jessie version already on GitHub.

Halen78, 2020-09-24:
> Sure. FYI you can always build a Jessie version yourself...just follow the instructions in the build.txt file in the directory. Alternatively if you don't want the latest release, there's a Jessie version already on GitHub.

@chundermike 
This is **awesome** news.  Thank you very much.  
I will pursue both options in the order you listed.
I'll close this issue, and submit a new one only if I get myself stuck again.
Thanks again!

closed 2020-09-24 by Halen78

Halen78, 2020-09-24:
@chundermike 
I followed the instructions in build.txt, and I was able, for the first time, to run fruitbox with no arguments and successfully pull up your help file, listing the command line arguments.  I'm very excited about this, and I'm going to dive into the manual and really get up and running.  Thank you very much for your help.

I'm embarrassed to say, I don't know how to make a pull request, but there are two very small changes I would like to suggest in your build.txt file.

Under the heading of:
> # install and build libmpg123 as static library...

There is a line:
> cd ~/mpg123-1.24.0

This command should not reference the home directory "~/".  Rather, based on the previous commands executed by the user, they should find themselves already in the parent directory of the mpg123-1.24.0 folder.  So the command should simply read:
> cd mpg123-1.24.0

Also, under the heading of:
> #touch support (jessie only?)...

there is a misspelling of "following" as "follwoing".

Thanks again for your help, and for releasing this awesome piece of software, I'm looking very forward to diving into it.

By the way, have you ever considered creating a video jukebox for the Pi?

chundermike, 2020-09-24:
Hi

Thanks!

I'll correct those typos in GitHub... thanks for pointing them out.

Fruitbox already has the ability to play videos... they need to be in ogv format though.