chundermike/rpi-fruitbox #48: UNC share for music

beit-no1, 2020-09-23:
I've setup my RaspBerry PI JukeBox as a dual boot with Pinnlite and it is great to have it swap between a JukeBox and a RecalBox arcade.  i did put two USB ports on the front so I can use these for the Xbox 360 controllers (if I don't go wireless).  

My Arcade JukeBox (aka FruitBox) is going to be network enabled, so was hoping to setup some method to allow a share location (on my synology NAS) to store the music and then through a restart it would just rebuild the music DB.

I have tried to setup the USB method so that a user could come and plug in a USB with a set of songs, but this seems very difficult between boots and to try and setup a key button to "reset" so that the jukebox rebuilds it's music.  This would allow us to have USB sticks with different music libraries, so a great concept, but I find linux just doesn't seem to make it easy to find a USB to mount.

Can you offer any good advice on either future shares and rebuild for the music.db (instead of deleting and rebooting) as well is there any fool-proof method of dealing with swapping USB's while the jukebox is operating (to shift from Rock to Country :))

Great software and is a wonderful fit into my mini-arcade (Virtual Pinball, Standup 2 player arcade, Cocktail cabinet + JukeBox).

beit-no1, 2020-09-23:
I should have also asked... Recalbox allows menu item to quit (reboot) and so I then get to Pinnlite menu and can select JukeBox OS, but is there an easy way in the jukebox to setup a "restart" type switch or menu item..  This would make it relatively easy to move back and forth and also load a new music DB as it could be deleted upon startup.

damanbaird, 2020-12-13:
UPnP/DLNA-support would be nice too.