chundermike/rpi-fruitbox #50: Please help: setup-noob

jcam2020, 2020-10-26:


chundermike, 2021-01-11:
Hi,

Sorry for slow reply.  fruitbox v2 will make setup a whole lot easier for you, so if you can be patient for a few more weeks....

jcam2020, 2021-01-12:
Hi! No problems!

That sounds great, I am already looking forward to this. I managed to setup a basic fruitbox using the image file . I found it too difficult to do without the image file. It’s a great software though and I am very grateful you created it. I am happy to send a small donation.

Thanks,

John.

Rasmusus, 2021-05-08:
can i ask u how? i too had the img. download, but i cant get  into any settings?

chundermike, 2021-05-08:
Hi,

To change settings, you need to edit the skin file (skins/.../fruitbox.cfg)

Mike

battlecat536, 2022-12-31:
Noob does not put it correctly as to how i struggle To change settings.
Older PI 2 .
I just can not get this to boot into modern screen. 
I'm either typing this command line very very wrong.

i can get to ;
 /rpi-fruitbox-master/skins/Modern $
anything after this no such file or directory.

help file i have read and still can not figure it out  :(

 you need to edit the skin file (skins/.../fruitbox.cfg)