chundermike/rpi-fruitbox #51: Cant find any MP3 Songs (Filepath Problem)

Smiagol, 2020-10-27:
Hey there,

im sitting in front of my Fuitbox for quiet some time now, and i can't find a Solution. 

My USB Drive is mounted to mnt/usb (but the directory is empty!?) and my "MusicPath" is /mnt/usb/   

On my drive is a directory called mp3 i had it up and running last year and it worked fine. i changed the Database Directory to my usb drive so i could use more than one drive without always waiting for the database creation. 

I hope someone can help me 

Smiagol

Smiagol, 2020-10-27:
just made another stick without any other Folders and it works...but every time i restart i have to select a Skin again

damanbaird, 2020-12-13:
i have all my music on one USB everytime i go to add music or take some off i just delete the .db file i believe it is and let it rebuild

chundermike, 2021-01-11:
fruitbox v2 (release coming soon) should make database handling simpler, so if you can wait a few weeks ;)