chundermike/rpi-fruitbox #54: Volume

tjveith, 2020-12-08:
Hello, is there a way to increase the volume going out through the headphone jack? I've attached a JBL speaker, and the volume is quite low at full volume. When I place different retropie cards in the raspberry pi, volume is brutally loud with retrogaming, so I know there isn't a problem with the jack itself. Any settings you could recommend to force an increase in volume with Fruitbox would be greatly appreciated. Thanks!

chundermike, 2021-01-11:
Volume on fruitbox is 0 to 100, adjustable on the - and = keyboard keys by default.  If the volume is still to low, try running alsamixer from the command line and increasing it there (if it's not max).

Cheers,
Mike

bigdrabz83, 2021-12-08:
Hi Mike.
Sorry to jump on this thread. I'm not skilled in any of this so please could you tell me if there is a way to increase the volume limit and if so how do I do it? 

Thanks

chundermike, 2021-12-08:
Hi

If fruitbox is set to maximum volume (using the volume buttons), and it's still not loud enough, it could be that the ALSA driver volume is not set to max. To change this, type alsamixer from the command line and change it with the up and down keys.

Mike

bigdrabz83, 2021-12-08:
I tried that, but it said there was no sound card.

bigdrabz83, 2022-02-02:
> Hi If fruitbox is set to maximum volume (using the volume buttons), and it's still not loud enough, it could be that the ALSA driver volume is not set to max. To change this, type alsamixer from the command line and change it with the up and down keys. Mike
> […](#)

I tried this but it just says that there is no soundcard. Any other ideas? 

Thanks,