chundermike/rpi-fruitbox #58: Not working correctly on a 1920 x 1080 Touch Screen

martinklein, 2021-01-04:
First of all Thousand Thanks for this wonderful Program, I love it !

I am on a 1920 x 1080 Touch Screen. All available Touch skins are for lower resolution and do not work correctly on my screen. When I touch on the top left of the screen I see that touch is recognized but only the first song starts playing.

Do I have to create a complete new skin for this higher resolution or is it enough to recalculate the touch areas ?

But the touch areas only define the buttons. How to modify the skin for song selection for touch for a higher resolution screen. I did not get the idea of how this works from reading your user guide. There are no touch areas for the [page] entries that define the songs.

With kind regards
Martin Klein

WernerDB456, 2021-01-04:
Brilliant software program ! Thank you very much.
I'm stuck with the same issue. I have an 19" IR Touch screen and none of the available drivers seems to work. Please help.....

chundermike, 2021-01-11:
Hi ,

Sorry that you're having problems. In theory, any size skin should work with any size touchscreen, as long as you've run the touch calibration using the --calibrate-touch command line option. Sounds like you both have.  In practice, there are lots of different touch screens on the market which have subtle differences in how they work (sometimes the touch resolution is different to the display resolution).  I've tried my hardest to cover all cases, but of course unless I have the touch screen itself to test with, it's quite often a shot in the dark.  One thing that would be useful is if you could run evtest and give me the report when you touch top left and bottom right corners of the screen...that will tell us the actual touch coordinates in your case.

For your information, all touch areas are specified relative to the skin size (specified in the skin cfg file).  fruitbox will scale these to match the touch screen size, so hopefully all touches will get translated to the correct skin coordinates.  Regarding the areas for the [page] entries, these are all calculated internally since fruitbox knows where the positions of the pages and corresponding song text is on the screen already.

Cheers,
Mike

martinklein, 2021-01-12:
I tried the "--calibrate-touch" but it just brought up the help text until I found that it must be used together with the "--config-buttons". So I did this now and in my fruitbox.btn it inserted:
TouchMin         = -2007 -2047
TouchMax         = 6105 6142

So I tried again the OneTouch and the WurlyTouch skins and now I can NEARLY select a song. There are most of the time near misses. So it either takes the song above the one I clicked or the one right of it. Seldom it picks the correct one.

But the Buttons I cannot click at all. Nothings happens.

Below is the result of running extest, selecting the screen and then touchhing top left and bottom right.

With kind regards. Martin

input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0xeef product 0xa107 version 0x210
Input device name: "eGalax Inc. eGalaxTouch EXC7910-1018-13.00.01"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      5
      Min        0
      Max     4095
      Resolution      13
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max     4095
      Resolution      24
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     4095
      Resolution      13
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     4095
      Resolution      24
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
  Event type 4 (EV_MSC)
    Event code 5 (MSC_TIMESTAMP)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1610457271.100442, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 244
Event: time 1610457271.100442, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 0
Event: time 1610457271.100442, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 54
Event: time 1610457271.100442, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1610457271.100442, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1610457271.100442, type 3 (EV_ABS), code 1 (ABS_Y), value 54
Event: time 1610457271.100442, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 0
Event: time 1610457271.100442, -------------- SYN_REPORT ------------
Event: time 1610457271.105503, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 9100
Event: time 1610457271.105503, -------------- SYN_REPORT ------------
Event: time 1610457271.118452, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 18100
Event: time 1610457271.118452, -------------- SYN_REPORT ------------
Event: time 1610457271.123462, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 27100
Event: time 1610457271.123462, -------------- SYN_REPORT ------------
Event: time 1610457271.136487, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 36200
Event: time 1610457271.136487, -------------- SYN_REPORT ------------
Event: time 1610457271.141378, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 45300
[... 139 lines cut ...]
Event: time 1610457273.737470, -------------- SYN_REPORT ------------
Event: time 1610457273.743439, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 63300
Event: time 1610457273.743439, -------------- SYN_REPORT ------------
Event: time 1610457273.755416, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 72400
Event: time 1610457273.755416, -------------- SYN_REPORT ------------
Event: time 1610457273.761427, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 81500
Event: time 1610457273.761427, -------------- SYN_REPORT ------------
Event: time 1610457273.773413, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 90600
Event: time 1610457273.773413, -------------- SYN_REPORT ------------
Event: time 1610457273.779381, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 99700
Event: time 1610457273.779381, -------------- SYN_REPORT ------------
Event: time 1610457273.791385, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 108700
Event: time 1610457273.791385, -------------- SYN_REPORT ------------
Event: time 1610457273.797370, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 117700
Event: time 1610457273.797370, -------------- SYN_REPORT ------------
Event: time 1610457273.810373, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 126800
Event: time 1610457273.810373, -------------- SYN_REPORT ------------
Event: time 1610457273.815378, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 135900
Event: time 1610457273.815378, -------------- SYN_REPORT ------------
Event: time 1610457273.828369, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 144900
Event: time 1610457273.828369, -------------- SYN_REPORT ------------
Event: time 1610457273.834354, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1610457273.834354, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1610457273.834354, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 153900
Event: time 1610457273.834354, -------------- SYN_REPORT ------------

martinklein, 2021-01-12:
Mike, I tested now in detail the touch behaviour of the skin iOtero/TouchWurly since this is the one I want to use.
There are Three Columns for the songs and in each Column are 6 Rows for the songs.
All Columns behave the same.
When I touch the song in row 1 nothing happens (only when I touch the very top of this box then the Song in Row 3 is selected)
When I touch the song in row 2 then the song in row 4 is selected
When I touch the song in row 3 then again the song in row 4 is selected
When I touch the song in row 4 then the song in row 5 is selected
When I touch the song in row 5 then nothing happens
When I touch the song in row 6 then the song of this row 6 is selected (hurray row 6 works !!)

I have to touch really in the middle of the box, when I press on the right side then the column to the right is selected in the same row schema as above.

And then as mentioned earlier - when I press any of the buttons nothing happens

With kind regards - Martin

chundermike, 2021-01-12:
Hi,

Try editing the fruitbox.btn file directly and change TouchMin to 0 0 and TouchMax to 4095 4095, and see if that helps.

Mike

martinklein, 2021-01-13:
YES ! Many thanks Mike. Now the song selection works 100% correct with touch.
Unfortunatly the buttons still dont work with touch. Nothing happens. They work only with the
associated keys...
Should I try with higher TouchMax value ?

Thanks again and regards. Martin

chundermike, 2021-01-13:
Hi,

Have you mapped the buttons as touch inputs? You'll need to run fruitbox with the --config-buttons command line option and then press the screen for the buttons you want to be touch sensitive.  For v2, this will be automatic :)

Mike

martinklein, 2021-01-14:
Mike, I did now the --config-buttons thing and everything works now.
MANY THANKS for all your help and your patience !
With kind regards. Martin

closed 2021-01-14 by martinklein

F925, 2023-07-15:
Good day. Not ben able to get my touch screen to work on Fruitbox. Read you posts and changed TouchMin to 0 0 and TouchMax to 4095 4095 This made some parts of the screen work but in the wrong aera. What would the correct settings for my 1920x 1080 Zeuslap portable monitor.