chundermike/rpi-fruitbox #60: Fruitbox 1.16 is this the latest version?

leewilks42, 2021-01-25:
Fruitbox 1.16 is this the latest version? I have seen that Mike is working on 2.0. I have a Pi 3B and a pi touchscreen and I can’t get the bottom buttons working

chundermike, 2021-02-15:
Yep, v1.16b. Have calibrated your touchscreen with the --calibrate-touch option?

leewilks42, 2021-02-15:
No what’s the command please? I will give it a whirl!

chundermike, 2021-02-15:
You need to use it with the config buttons option, so it'll  be...

./fruitbox --calibrate-touch --config-buttons

Mike

tarbard, 2021-03-10:
> Fruitbox 1.16 is this the latest version? I have seen that Mike is working on 2.0. I have a Pi 3B and a pi touchscreen and I can’t get the bottom buttons working

Did you get this to work?   I am also using a 3B with an official touchscreen,  I can press song names and they are added to the list but the other buttons don't respond. There is no feedback after touching any button on the screen. Sometimes after 30 seconds or so after touching the "next page" button it will scroll to the next page. 

I have calibrated my screen and confirmed the skin has touch areas defined (i'm using TouchBlue but I have the same result with every skin).