chundermike/rpi-fruitbox #61: No sound

bigdrabz83, 2021-02-10:
Hi Mike. I'm having trouble getting sound to play. This will be the 4th time I have installed fruitbox. Each time the sound played great. Now after a fresh install and installing a Pirate Audio 3w amp to my RPi 3B+ there is no sound playing through my TV or the speakers I wired to the amp. Any suggestions? 

Thanks. Keep up the good work.

schnho, 2021-02-22:
@bigdrabz83 I'm going to stick my neck out, as I haven't got anyway to test this, but I too do not have sound after doing a raspbian update.

If you install mpg123 (v1.25.0), sound plays out of the pi. I noticed that the mpg123 libraries are packaged with fruitbox 1.16b is 1.24.0.

Maybe fruitbox needs to be built with latest mpg123 libraries for sound to come back.

My work around (I'm not proud) is by NOT updating the pi when using raspbian 2021-01-11 img.

bigdrabz83, 2021-02-22:
> @bigdrabz83 I'm going to stick my neck out, as I haven't got anyway to test this, but I too do not have sound after doing a raspbian update.
>
> If you install mpg123 (v1.25.0), sound plays out of the [...]

I manager to solve the issue by changing the dtparam im boot/config.txt

The instructions are in this link 
https://radio-docs.netlify.app/docs-audio

I hope that helps

schnho, 2021-03-04:
Thanks for the fix.

As I don't have a hifiberry and only using the 3.5mm jack with an HDMI display (heathen, I know) I fixed my issue by setting the default output to the headphone socket by adding file to /etc/asound.conf with the lines
`   defaults.pcm.card 1`
`   defaults.ctl.card 1`
I'm guessing fruitbox plays though the default soundcard.