chundermike/rpi-fruitbox #64: Can not get it to launch from Desktop shortcut

mdyeargin, 2021-03-04:
Using the following to do short cut:

nano ~/Desktop/myshortcut.desktop

[Desktop Entry]
Version=1.1
Type=Application
Name=FruitBox
Comment=Juke Box Launch
Path=/home/pi/rpi-fruitbox-master
Exec=./fruitbox --cfg skins/TouchOne/fruitbox.cfg
Terminal=true
Icon=myicon.svg

YEILDS ERROR!!!!!

Invalid desktop entry file: '/home/pi/Desktop/myshortcut.desktop'

When run in terminal from rpi-fruitbox-master directory using rpi-fruitbox-master
to get pi@raspberrypi:~/rpi-fruitbox-master $ prompt the following command is run

./fruitbox --cfg skins/TouchOne/fruitbox.cfg

No issues starting and playing.

aavolcanoaa, 2021-03-14:
I suspect you don't have enough memory allocated in sudo config check its set at 256

mdyeargin, 2021-03-15:
That assumption would be incorrect. As stated if run from the terminal line the application runs with no issue, it is only when trying to create a desktop short cut to perform  a click launch. The 'instructions were followed and the GPU 256 was actually set prior to running the install.sh. 

To reiterate, it is only creating a desktop launch which I am having the issue.

aavolcanoaa, 2021-03-16:
Only trying to help.   I’m out