chundermike/rpi-fruitbox #65: ELO Touchscreen Monitor not responding

mjvracing, 2021-03-06:
Hi there,

I came across this software after looking for some to use as a jukebox for my upcoming wedding. Looks amazing!

I have installed it on a RPi3 and have had to run it from the desktop UI to get the 800x600 resolution I needed to work - may look into this later to run it from command line.

Because the ELO monitor appears to just have mouse positioning sensors, rather than multitouch or similar, the touch screen calibration does not work, and config-buttons just suggests a left mouse click every time I tap. Is there a way around this, or do I need to get a more modern touch panel?

Also, while I am here, are there any options to add crossfade between tracks? I've got the system on autoplay with no duplicates and a song history list long enough to prevent replays for the wedding, but it would be great if crossfade was also available.

Looking forward to seeing what v2.0 brings

schnho, 2021-03-11:
As the config-buttons setup is registering your touch panel as mouse clicks, I would suggest you would not need the fruitbox.btn file as the panel works like a mouse. Unfortunately I haven't run fruitbox in a desktop environment, so I don't know if you can use a mouse to select tracks.

chundermike, 2021-05-01:
v2 has a fix for the ELO touchscreen problem

closed 2021-05-20 by chundermike