chundermike/rpi-fruitbox #67: Button configuration

bigdrabz83, 2021-03-12:
I have just installed fruitbox using the SD image. Boots automatically from start up. Great! The problem I now have is that I cabt figure out how to configure my buttons. After I select a skin, the MP3's load and then that's it. When I press esc, everything reboots and I'm back where I started. Please can someone tell me how I can configure my buttons? 

Thanks

schnho, 2021-03-12:
If you shut down fruitbox and get to a command prompt. 

Go into the fruitbox directory.

`cd /home/pi/rpi-fruitbox-master`

Run fruitbox with flag --cfg-buttons

`./fruitbox --cfg-buttons`

Follow on screen instructions. If you want touchscreen to be the buttons, just touch the screen.

This is all documented really well in the read.me and the userguide files.

bigdrabz83, 2021-03-12:
I couldn't get to the command prompt at all. That was the issue. I managed to stumble across a way to do it (i think it was ctrl + esc). Everything is working great now.

Thanks