chundermike/rpi-fruitbox #68: Using a Raspberry Pi 7 in Screen (Official) not working

mdyeargin, 2021-03-15:
Using a Raspberry 3 B+ and a Raspberry Pi 7" touch screen, I put the full image using the SD image file is available. If I use a full Buster image, I have no issues with screen. Using the SD image I get no response. I tried to load the second option in the instructions found "Rather than use the SD card image" and have no issues with Fruitbox running.

Any assistance in how to get Fruitbox to recognize the 7" touch ability by default would be great.

lloyd61, 2021-03-24:
Same here, my touch screen works fine with Raspberry OS, but whilst it is recognised when Fruitbox boots up, it will not respond to any touch. Please help.

mdyeargin, 2021-03-24:
Was able to find link to get all but ‘Random’ to work. If needed I can post.

lloyd61, 2021-03-25:
Yes please

schnho, 2021-03-25:
If it works with Raspberry OS, why not use it? The SD image will probably be older versions of both os and fruitbox.

Chuffy2020, 2021-04-30:
> Was able to find link to get all but ‘Random’ to work. If needed I can post. 

Would you mind posting the link?  Thanks.

argyl3gargoyl3, 2021-05-04:
I too am having trouble with the image and getting the touch screen to respond.   Using a EVICIV Raspberry Pi 10.1" Touch Screen Monitor with Case, thanks.

lloyd61, 2021-05-04:
In the end I gave up, other battles to deal with.

Chuffy2020, 2021-05-04:
Sounds like V2.0 might have a fix for this issue.  On standby..... :-)

mdyeargin, 2021-05-04:
Here is what I finally tried. The 7" will auto start in random mode and allow you to select next song:
Using the following to do short cut:

nano ~/Desktop/myshortcut.desktop

[Desktop Entry]
Version=1.1
Type=Application
Name=FruitBox
Comment=Juke Box Launch
Path=/home/pi/rpi-fruitbox-master
Exec=./fruitbox --cfg skins/TouchOne/fruitbox.cfg
Terminal=true
Icon=myicon.svg

YEILDS ERROR!!!!!

When run in terminal from rpi-fruitbox-master directory using rpi-fruitbox-master
to get pi@raspberrypi:~/rpi-fruitbox-master $ prompt the following command is run

./fruitbox --cfg skins/TouchOne/fruitbox.cfg

Result in fruitbox.log is as follows with no issue starting:

fruitbox v1.16b (Jun  9 2020, 20:32:27) by Mike Kingsley
A customisable MP3 Retro Jukebox for the Raspberry Pi.
Detecting display...1824 x 984 (60Hz)
NOTE: Opened "NOVATEK USB Keyboard" (/dev/input/event0)
NOTE: Opened "NOVATEK USB Keyboard System Control" (/dev/input/event1)
NOTE: Opened "NOVATEK USB Keyboard Consumer Control" (/dev/input/event2)
NOTE: Opened "YSPRINGTECH USB OPTICAL MOUSE" (/dev/input/event3)
Reading configuration 'skins/TouchOne/fruitbox.cfg'...

  [general]
  SkinName = TouchSong
  SkinDescription = Touch skin for Official Raspberry Pi 7" Touch Screen
  SkinSize = 800 480
  Database = ../fruitbox.db
  MusicPath = ../Music/
  SortSongsBy = Title
  SortSongsBy = Artist
  AutoSelect = true
  SelectTimeout = 150
  SelectHoldTimeout = 150
  AutoPageTurnTime = 3000
  MaxPlaylistLength = 20
  LoopPlaylist = false
  PlaysPerCoin1 = 0
  SongsPerPage = 6
  PageSize = 360 308
  PairSongs = Dual
  ArtistDescription = Artist
  ArtistDescription = " / "
  ArtistDescription = PairedArtist
  PageMoveStyle = 20 0 false
  SongLoadTime = 0
  SongUnloadTime = 0
  ArtistText = 1 255 32 32 255 centre true true false 0 7 294
  SongText = 0 0 0 0 255 centre true true true 0 7 360

  [sounds]

  [font]
  File = skins/TouchOne/bluecond.ttf
  Height = 24

  [font]
  File = skins/TouchOne/bluecond.ttf
  Height = 22

  [font]
  File = skins/TouchOne/dotmatrx.ttf
  Height = 16

  [font]
  File = skins/TouchOne/bluecond.ttf
  Height = 18

  [bitmap]
  File = skins/TouchOne/background.jpg
Position = 0 0
  Size = 800 480

  [page]
  Position = 30 100
  Background = skins/TouchOne/strip3.jpg

  [page]
  Position = 410 100
  Background = skins/TouchOne/strip3.jpg

  [status]
  Position = 30 12
  Size = 352 84
  Text = 2 255 255 255 255 right false true false 0 0 350
  Contents = skins/TouchOne/now_playing.txt
  LineSpacing = 4
  TimerTickPeriod = 60

  [status]
  Position = 414 12
  Size = 352 84
  Text = 2 255 255 255 255 left false true false 0 0 350
  Contents = skins/TouchOne/coming_up.txt
  LineSpacing = 4

  [status]
  Position = 160 432
  Size = 234 53
  Text = 2 255 255 255 255 centre false false false 0 0 350
  Contents = skins/TouchOne/help.txt
  LineSpacing = 4
  TimerTickPeriod = 200

  [touch song]
  Bitmap = skins/TouchOne/touch_red.png

  [bitmap]
  File = skins/TouchOne/foreground.png
  Position = 0 0
  Size = 800 480

  [bitmap]
  File = skins/TouchOne/volume.png
  Position = 8 422
  Size = 12 51
  VertScale = Volume
  VertScaleMode = Clipped

  [status]
  Position = 423 421
  Size = 93 53
  Bitmap = skins/TouchOne/random_button.png
  Contents = skins/TouchOne/random_pressed.txt
Contents = skins/TouchOne/random_pressed.txt

  [status]
  Position = 546 421
  Size = 94 53
  Bitmap = skins/TouchOne/skip_button.png
  Contents = skins/TouchOne/skip_pressed.txt

  [status]
  Position = 663 421
  Size = 54 53
  Bitmap = skins/TouchOne/left_button.png
  Contents = skins/TouchOne/left_pressed.txt

  [status]
  Position = 37 421
  Size = 93 53
  Bitmap = skins/TouchOne/volume_button.png
  Contents = skins/TouchOne/volume_pressed.txt

  [status]
  Position = 717 421
  Size = 53 53
  Bitmap = skins/TouchOne/right_button.png
  Contents = skins/TouchOne/right_pressed.txt

  [touch areas]
  ButtonRandom = 423 421 94 53
  ButtonSkip = 546 421 94 53
  ButtonLeft = 663 421 48 53
  ButtonRight = 724 421 45 53
  ButtonVol- = 38 421 32 53
  ButtonVol+ = 98 421 32 53

Initialising audio ...
NOTE: Database out of date, will regenerate it.
Database '../fruitbox.db' not found, so will create it for you...
Creating new database '../fruitbox.db'...
This may take some time, but only needs to be done once.
If your music library changes please delete '../fruitbox.db' and re-run fruitbox.

Alternatively, if you are feeling brave, you can edit the database file directly,
using a text editor.  This can be useful if you want to add or remove songs manually,
and/or modify their tag data (song title, artist name, etc.)

Searching ../Music/ ...

Loading database '../fruitbox.db'...
1519 songs found
Sorting songs by Title...
Sorting songs by Artist...
Building song pages...
Let's Rock!

lloyd61, 2021-05-05:
Ok, thank you.