chundermike/rpi-fruitbox #69: Adding muliple UBS flash drives to fruitbox

bigdrabz83, 2021-03-22:
I was wondering if someone could walk me through adding multiple flash drives to fruitbox so I can quickly add a "Top 40" or a "Christmas Songs" drive instead of adding and deleting songs from the main flash drive if it is even possible. 

Thank you

Witchboard, 2021-03-22:
You could just make stand alone USB drives that include fruitbox, the skin and the songs.  Otherwise, you can just add multiple music paths to your skin configuration.

bigdrabz83, 2021-03-22:
Thank you. Would you be able to tell me how to add the multiple paths please? I'm really new to all this coding malarky and don't want to mess anything up. As I'm using the latest image fruitbox found my flash drive automatically. I wouldn't know where to start adding the code needed to do the same.

Witchboard, 2021-03-22:
You just edit the configuration file of the skin you're using and include multiple paths.  There's already one in there for an example, just make as many as you need, one per line with all your paths and it will scrape everything into the database.

bigdrabz83, 2021-03-22:
Excellent. I think I know enough to give that a go. Thanks for the help.