chundermike/rpi-fruitbox #71: Feature request: Ignore "the" when sorting by artist

AtebMT, 2021-04-08:
It's a small issue, but I have my tracks sorted by artist, but that does mean "The Beatles" comes under T, rather than B which is where I would like it. It means I get a ton of "The ..." artist names.

It's not a big deal, but having an option to ignore "the" would be very useful for me!

DaveUltra, 2021-04-09:
Hey! I'm a new contributor here and I'd like to help with this "issue". I think sorting names by ignoring the "The" isn't that hard and I would like to work on this.

AtebMT, 2021-04-12:
My work around is to (manually right now)  post process the `fruitbox.db` file and update the artist name there. My next step is to automate that process. A proper code solution would be better though.

chundermike, 2021-05-01:
This feature is now been implemented and it will be available in release v2

AtebMT, 2021-05-03:
That's awesome. Thanks.

closed 2021-05-20 by chundermike