chundermike/rpi-fruitbox #72: errors

suzaru56, 2021-04-09:
ERROR: Couldn't open configuration file 'skins/wurly/fruitbox.cfg' for input I have received several errors while tryhing to run fruitbox. 
Ones is the error above. I have had one where it said it couldn't find any mp3 files. I have attempted to change the path in the config file.....even though it looks like it is pointing to the correct path. Then I realized the files I tranfered to the pi were actually m4a files so I deleted those and added only mp3 files. that is when I received the error above.
I really apprectiate your help!

suzaru56, 2021-04-09:
ERROR: Can't find any MP3 songs (are the 'MusicPath' settings in your config correct?)
I tried to run a different skin to see if when I tried to change the music path in wurly might have messed something up. The path in wurly I changed to home/pi/music. 
When I looked at the config file for WallJuke the music path says ../Music/

Witchboard, 2021-04-09:
What's the path for your music?  Remember, Linux is case sensitive.  If you mapped to "music", but the directory is "Music", then it's not going to find it.  The path need to be in relation from the rpi-fruitbox-master directory.  So if you're music path is home/pi/music, then ../music should be working.

suzaru56, 2021-04-09:
Well, I have tried two places. home/pi/Music and
home/pi/rpi-fruitbox-master/Music I get the same error ERROR: Can't find
any MP3 songs (are the 'MusicPath' settings in your config correct?) I read
something about deleting the fruitbox.db file.....do I need to do that and
if so how and where do I do that?
*Thanks*
Susan

schnho, 2021-04-09:
Are you starting fruitbox from terminal?

If you change the fruitbox.cfg in Wurly skin to absolute file location, this means it shouldn't matter if you download fruitbox into a different directory.

`/home/pi/Music`

cd to where your fruitbox program, then start fruitbox
`.fruitbox --cfg skins/Wurly/fruitbox.cfg`

suzaru56, 2021-04-09:
yes that is the way it is set up and I still get the ERROR: Can't find
any MP3 songs (are the 'MusicPath' settings in your config correct?
Could there be something wrong with how mu mp3 files are named? The have numbers in front of them ,,,not just the name of the fiile like" 01 Against The Grain.mp3' not 'Against The Grain.mp3'

suzaru56, 2021-04-09:
I performed a reboot and trie to run it again and this time received this errorERROR: Couldn't open configuration file 'skins/Wurly/fruit.cfg' for input

schnho, 2021-04-09:
Unless you created your own config file in the Wurly directory, the file is called fruitbox.cfg

suzaru56, 2021-04-09:
OH Geez I typed it wrong.....I will try again! 
Thanks

suzaru56, 2021-04-09:
I typed it correctly this time and received this error: ERROR: Can't find any MP3 songs (are the 'MusicPath' settings in your config correct?)

suzaru56, 2021-04-09:
I changed my mp3s to not have a number in the title. They had numbers in front of them ..not just the name of the file like" 01 Against The Grain.mp3' so I changed them to 'Against The Grain.mp3 'with no number in front.  I verified that my config file had home/pi/Music. Then reran the command. It didn't give me errors this time but it looked like it was going to work then my screen went black and froze everything up. I have to unplug to reboot the pi.
Any ideas
Thanks

Witchboard, 2021-04-09:
Can you attach your fruitbox.cfg file from your Wurly directory?

suzaru56, 2021-04-09:
You bet
[image: image.png]
Thanks

suzaru56, 2021-04-10:
Hi
I was wondering if you received my image of fruitbox.cfg file. I am afraid
it did not email well and I need to send it in a different format.
Please let me know.
Susan

schnho, 2021-04-10:
No image :( can you cut and paste the text instead?

suzaru56, 2021-04-10:
Sure!

[general]
SkinName = Wurly
SkinDescription = by chundermike
SkinSize = 1920 1080
Database = ../fruitbox.db
MusicPath = /home/pi/Music
SortSongsBy Title
SortSongsBy Artist
SelectButtons = ABCDE
SelectButtons = 1234
AutoSelect = Yes
SelectTimeout = 150
SelectHoldTimeout = 150
AutoPageTurnTime = 3000
MaxPlaylistLength = 20
LoopPlaylist = yes
AutoPlay = yes
AutoPlayGap = 3000

THanks!
Susan

schnho, 2021-04-10:
Maybe the id3 tags on your mp3s need tidying up?

schnho, 2021-04-10:
Also what hardware are you using? Rpi4 is not supported.

ZS6TVB, 2021-04-10:
Does :

ls -lr /home/pi/Music 

Show your MP3 files?

suzaru56, 2021-04-10:
How do I clean up the id3 tags?
Yeh, I started out trying to use the RPI 4 and found out that I couldn't use that one so I just bought a Pi 3b. 
Thanks
Susan

schnho, 2021-04-10:
Depending how you got your mp3s will depend on how the id3 tags were created.

You can edit them with software like vlc.

suzaru56, 2021-04-10:
Some I purchased on ITunes but they seem to be m4a's and then when I purchased the cd's I ripped onto my computer but do not remember what software I used to rip them . I did all that a long time ago
Susan

suzaru56, 2021-04-10:
How did you know that they had id3 tags?

schnho, 2021-04-10:
I think fruitbox uses the id3 tags to display song title and artist name. Without it I think it fruibox skips the track. If you don't have id3 tags it will probably not find any tracks to put into the database.

If you can get the mp3 files on another computer, play the track. If the song name artist album is displayed the id3 tag is probably correct. If it just shows the filename, the id3 tag is probably missing.

suzaru56, 2021-04-10:
Ah...it seems as if the id3 info is on my music becase the name of the track and the name of the artist is present when I play the song on another player. I do find this though ....encoded by iTunes v7.0.2.16 ...could this be a problem?

suzaru56, 2021-04-10:
OK, I re-ripped a CD with Windows media player and when I run Fruitbox the fruitbox lable comes up and flips through all the 11 songs I have on the pi. but then the screen goes black and does not display the songs for me to choose. I am using a 7 inch touchscreen. Also, entering  ls-lr /home/pi music shows that I have the songs in that directory.

suzaru56, 2021-04-10:
I just tried the same as above without the touchscreen and received the same results. I read somwhere about deleting fruitbox.db and rerunning the fruitbox software. Do I need to do that and if so how do I do that?

schnho, 2021-04-10:
According to your config file the fruitbox.db should be located one level up from where you run fruitbox from.

`cd ..
rm fruibox.db`

suzaru56, 2021-04-10:
So, I removed fruitbox.db then reran fruitbox....it came up and showed it re-creating the database and then the screen starts to dim and continues until the screen goes black.
This is for a school project due on April 22nd. Should I look for another software? I really like this software but I don't have tons of time to get this figured out. Do you think we can get it figured out?
I appreciate your opinion!
Thanks 
Susan

schnho, 2021-04-10:
Are you running raspberrypi OS with a desktop environment or are you booting into command line?

Did you download the fruitbox image which boots straight into fruitbox from startup or did you download from the files from github?

suzaru56, 2021-04-10:
I am running rpi OS with a desktop environment but I am running fruitbox via a command line.
I downloaded the image from github
On my fruitbox it does not boot straight to fruitbox. I have the Rpi OS and the Fruitbox on the same sdcard

suzaru56, 2021-04-10:
I used these instructions
It is recommended to start from a fresh OS install (latest Raspbian Buster release)
Increase GPU memory to (at least) 256 (sudo raspi-config -> Advanced Options -> Memory Split)
From a command shell, type : wget https://github.com/chundermike/rpi-fruitbox/raw/master/install.sh
From a command shell, type : chmod +x ./install.sh
From a command shell, type : source ./install.sh
Copy your music MP3 files (either to the SD card or USB memory stick)
Point fruitbox to your MP3 files (edit skins/WallJuke/fruitbox.cfg (or any other skin you fancy) and change the MusicPath parameter)
Run fruitbox ( ./fruitbox --cfg skins/WallJuke/fruitbox.cfg)

schnho, 2021-04-10:
Try running without desktop.

Open terminal

`sudo raspi-config`
Choose 1. System option
Choose S5 Boot / Auto....
Choose B2 Console Autologin

Then run fruitbox as you did before.

suzaru56, 2021-04-10:
OK, I made thses changes and it booted straight to a terminal. I reran the Fruitbox......the logo came up the it fades to black as it has been doing

schnho, 2021-04-10:
The last think i can suggest is that you try some other skins. After that I'm out of ideas.

suzaru56, 2021-04-10:
So ,I tried Walljuke and received all of this info
pi@raspberrypi:~/rpi-fruitbox-master $ ./fruitbox --cfgskins/Walljuke/fruitbox.cfg

fruitbox v1.16b (Jun  9 2020, 20:32:27) by Mike Kingsley
A customisable MP3 Retro Jukebox for the Raspberry Pi.

Command line arguments...

--cfg <filename>           : select (skin) configuration file
--database <filename>      : select database file (overrides skin cfg Database)
--music-path <filepath>    : select music path (overrides skin cfg MusicPath)
--choose-cfg <filename>    : choose a skin using the menu defined in <filename>
--user-guide               : creates the user guide ("userguide.txt" file)
--save-playlist <filename> : on exit, save the current playlist to <filename>
--load-playlist <filename> : load the playlist from <filename>
--debug-song-nums          : show song/albums numbers next to songs to help identify them
--no-database-update       : Disable update of database if out of date
--config-buttons           : configure button mappings
--calibrate-touch          : calibrate touch screen
--test-buttons             : test button codes
--screenshot               : save a screenshot of the skin
--no-scale                 : Don't scale the skin to fit the display
--button-map <filename>    : Button mapping file (default fruitbox.btn)

Acknowledgments...

Allegro v5.2.4[1] Game Programming library for Graphics and Sound.
libmpg123 v1.24.0 for MP3 decoding.
DWJukebox for inspiration!

To encourage future development, please consider making a donation
to the author at https://paypal.me/rpifruitbox

schnho, 2021-04-10:
That's because you missed a space between --cfg skins....

suzaru56, 2021-04-10:
Fixed that an then received this error
ERROR: Couldn't open configuration file 'skins/Walljuke/fruitbox.cfg' for input

schnho, 2021-04-10:
WallJuke (capital J)

suzaru56, 2021-04-10:
Fixed all that still faded to Black

schnho, 2021-04-10:
After it fades to black, if you ctrl c are their any error commands to give you any clues?

suzaru56, 2021-04-10:
No, it seems to freeze everthing up and I have to turn off the pi and turn it back on to reboot it

schnho, 2021-04-10:
I'm afraid without any logging, i'm out of ideas.

suzaru56, 2021-04-10:
OK, Thank you sooo much! I appreciate it even though I didn't get it going!
Susan

closed 2021-04-10 by suzaru56