chundermike/rpi-fruitbox #73: TouchOne  skin works!

suzaru56, 2021-04-10:
OMG! I tried TouchOne skin and it worked YIPPEE! 
But now I am having trouble getting the buttons to work.
Any help is appreciated!

schnho, 2021-04-11:
You need to configure the buttons before running fruitbox.

At the terminal

`./fruitbox --configure-buttons`

Fruitbox will start up but will prompt for how you want to control fruitbox. If you want that function to be touchscreen, press anywhere on the touchscreen.

Once it has gone through all the controls, it will quit out.

Restart fruitbox.

suzaru56, 2021-04-11:
Do I need to write it this way so it knows what skin I am mapping the buttons on?
./fruitbox --configure-buttons/TouchOne/fruitbox.cfg
When I wrote./fruitbox --configure-buttons I received this info
Command line arguments...

--cfg <filename>           : select (skin) configuration file
--database <filename>      : select database file (overrides skin cfg Database)
--music-path <filepath>    : select music path (overrides skin cfg MusicPath)
--choose-cfg <filename>    : choose a skin using the menu defined in <filename>
--user-guide               : creates the user guide ("userguide.txt" file)
--save-playlist <filename> : on exit, save the current playlist to <filename>
--load-playlist <filename> : load the playlist from <filename>
--debug-song-nums          : show song/albums numbers next to songs to help identify them
--no-database-update       : Disable update of database if out of date
--config-buttons           : configure button mappings
--calibrate-touch          : calibrate touch screen
--test-buttons             : test button codes
--screenshot               : save a screenshot of the skin
--no-scale                 : Don't scale the skin to fit the display
--button-map <filename>    : Button mapping file (default fruitbox.btn)

Acknowledgments...

Allegro v5.2.4[1] Game Programming library for Graphics and Sound.
libmpg123 v1.24.0 for MP3 decoding.
DWJukebox for inspiration!

To encourage future development, please consider making a donation
to the author at https://paypal.me/rpifruitbox

suzaru56, 2021-04-11:
Never mnd....I figured it out! Thanks
I was wanting to try out the iOtero Skins .....would I enter .fruitbox --cfg/iOtero/TouchWurly/fruitbox.cfg?
THanks

suzaru56, 2021-04-12:
Hi there I am not getting any sound from fruitbox.....if I just play the audio file via VLM I can hear the music but not through fruitbox.
Any help appreciated

schnho, 2021-04-13:
See issue #61, no sound.

suzaru56, 2021-04-14:
Thanks...I have sound now.
I appreciate your help!
Susan

closed 2021-05-20 by chundermike