chundermike/rpi-fruitbox #75: Touchscreen

Chuffy2020, 2021-04-27:
I know I must have missed something, but I've been through the user guide and readme and can't for the life of me figure this out.  I have bought a Waveshare 1024×600, 10.1 inch Capacitive Touch LCD Screen.  When I do ./fruitbox --config-buttons, each button is registered as left mouse click.  I have then gone into fruitbox.btn and changed them all to touch, but still nothing.  What am I doing wrong!?   I have seen in some forums and queries that some users enter their TouchDevice at the start of the fruitbox.btn file but I haven't actually seen this in an instruction anywhere.  Did I miss it?  Any help greatly appreciated as I am so close to the end of my project.  Thanks.  Chuffy

Chuffy2020, 2021-04-27:
To add, when I run 'sudo ./fruitbox --config-buttons -calibrate-touch' it comes up to press top left target but the screen is unresponsive.  Maybe it's just this screen model.  I saw in the forum a few others having issues with it.

chundermike, 2021-05-01:
Hi Chuffy,

Maybe it's the same problem I've had with ELO touchscreens, which work slightly differently to the one I have (official Pi 7" touchscreen). I've got the ELO touchscreen support in fruitbox working now but it won't be released until v2 comes out

Chuffy2020, 2021-05-02:
Thanks for getting back to me, Mike.  Much appreciated.  No probs - will wait for V2 to come and fingers crossed that fixes it.  Really enjoying fruitbox in my test set up - just hoping I can get the touch screen to work on my project.  Keep up the good work - great, fun piece of software.