chundermike/rpi-fruitbox #76: Newbie Getting Started

stricko1957, 2021-05-06:
Hi there. Despite having worked in IT for 30 years, I'm new to Linux, so be kind if I ask stupid questions. I'm using the SD card approach on a PI 3b, and have sort of found my way around. I'm trying to get the Pi to boot to a specific skin, so as per the instructions I've edited (using Nano) the rc.local file in /etc to point to the correct fruitbox.cfg file. But when I try to save the file, I get an error message telling me "Permission Denied".

When I use Esc to leave Fruitbox, I'm asked to Login, and in the absence of any other info, I'm signed on as the default "Pi" user. Should I be signed on as something else to get the required permissions. Many thanks

stricko1957, 2021-05-06:
Ignore thae last request, I know discovered what sudo does........ . Next problem is getting the Touchscreen to work. I've run the -- config-buttons option, but the screen doesn't seem to be responding correctly. I've got a couple of areas active on the screen, but not all. Stupid question I know, but during button config, if you want Touch for a specific button, do you have to touch the actual button area on the screen, or just anywhere?

chundermike, 2021-05-08:
Hi,

During config-buttons, just touch anywhere on the screen to register that button as touch.  Regarding your touchscreen, there is a problem with ELO touchscreens (i.e. they don't work with fruitbox), so maybe it's a similar issue.  I've fixed this bug now and it'll be released in v2, so hopefully that'll fix your problem at the same time.  Also in v2 touchscreen button setup is automatic, so that'll be a lot less painful too.

Cheers,
Mike

chundermike, 2021-05-08:
> Thanks Mike. $64,000 question: when do you think V2 might be released. Happy to help if you need any beta testing.

Hi,

I'm pushing for release at the end of this month. All code changes are done,  just debugging now, then I need to convert a few skins to work with v2. Thanks for the offer to beta test; I'll release ASAP then would appreciate any feedback after that :)

Cheers
Mike