chundermike/rpi-fruitbox #79: "Failed to add service - already in use" error on pi4 (retropie setup)

mr-falken, 2021-05-16:
Hi,
 I'm trying to add fruitbox to my pi4 powered bartop, but got this error as soon as I try to start fruitbox: is that something related to pi4 specifically ?

 Kernel version 4.19.97-v7l+

Witchboard, 2021-05-17:
From the readme
> fruitbox works on all Raspberry Pi devices except model 4

I think I read where Pi4 will be supported in version 2.0 when it's released.

chundermike, 2021-06-07:
> From the readme
> 
> > fruitbox works on all Raspberry Pi devices except model 4
> 
> I think I read where Pi4 will be supported in version 2.0 when it's released.

Yes, that's correct.  The Allegro 5 library I'm using doesn't currently work on Pi4 with Buster Lite, although the dev guys are trying to sort this out atm. In the meantime, when v2 comes out I'll also release a Buster Desktop version which will run on Pi4.