chundermike/rpi-fruitbox #80: V2?

stricko1957, 2021-06-07:
Hi there, how's progress on V2?

batfink5000, 2021-06-12:
Hi.  

Im also interested in V2, i have an ELO touchscreen on an rpi4 for this project.
If you can share anything around the likely release date, that would be great.

Thanks.
BF

chundermike, 2021-06-30:
All changes are done, testing atm. House move has meant progress has slowed down. Looking like release at the end of July now...

stricko1957, 2021-06-30:
Many thanks

sergiohinniger, 2021-09-03:
Cant wait! Any news about the release date?

chundermike, 2021-09-07:
Sorry, still wouldn't like to commit to a release date atm. Defo before Christmas!

rpt007, 2021-09-12:
Mike, pls take your time.
House and family first, fruitbox second. Quality should be the main focus. 
For those who can't wait - get a Raspi 3B+ and you are up and enjoying this excellent piece of software.
This will bridge the time until an even better solution is out.

batfink5000, 2021-12-02:
Hey there.
Any news on a release date for V2 ?

Thanks BF

chundermike, 2021-12-02:
Hi,

Hopefully sometime in December, maybe a couple of weeks time.

Cheers
Mike

batfink5000, 2022-02-18:
Hi Mike.

How are things looking for V2 ?

I appreciate the effort involved in supporting projects such as these, priorities change and were often not able to donate the time needed.  With that in mind, is there any possibility that you could put up the v2 code on a dev or beta branch ?
Im really stuck without rpi4 support.  

Thanks BF