chundermike/rpi-fruitbox #81: Music Path Issue

Wookie922, 2021-06-14:
This is a me issue, I have 0 intuitive linux knowledge.  I can not seem to get the correct path for my music files.  They are located in the Music folder under rpi-fruitbox-master.  I have configured  the skin "WallJuke" in many different ways with no luck.  Maybe I need to delete the database folder? I'm not sure where that is located to delete it.   Every time start a new project like this it teaches me humility.

Wookie922, 2021-06-14:
I think I did something right. It appears to be creating the database. Or it's frozen. How long should that take? It's been a couple of hours. I have 1000 songs in the file [image]but that is only a fraction of my collection.

Wookie922, 2021-06-14:
> This is a me issue, I have 0 intuitive linux knowledge.  I can not seem to get the correct path for my music files.  They are located in the Music folder under rpi-fruitbox-master.  I have [...]

Witchboard, 2021-06-15:
The database is created when your directories are scanned.  If it's not picking up on the directory, then it's not going to create a database.

Post your config file and we can look and see the directory structure.

Wookie922, 2021-06-15:
I've tried this many different ways but this is current. I copied the path I used to put the files in the folder.

[general]
SkinName = WallJuke
SkinSize = 1024 768
Database = ../fruitbox.db
MusicPath = ~/rpi-fruitbox-master/Music
SortSongsBy = Title
SortSongsBy = Artist
SelectButtons = ABCDE
SelectButtons = 1234
AutoSelect = Yes
SelectTimeout = 150
SelectHoldTimeout = 150
AutoPageTurnTime = 3000
MaxPlaylistLength = 20
LoopPlaylist = no
AutoPlay = yes
AutoPlayGap = 10
PlaysPerCoin1 = 0
SongsPerPage = 2

Witchboard, 2021-06-15:
Maybe it's my own ignorance with Linux, but I don't know what the tilde means in a file path.

Wookie922, 2021-06-15:
I know nothing about Linux but from the boot menu down it has them. And north of the boot menu does not. No idea why and if it needs to be there. I tried it without it as well.

Wookie922, 2021-06-15:
here is where the music files are 

pi@JukeBox:~/rpi-fruitbox-master/Music $ ls
 17.mp3
 1999.mp3
'2 Minutes to Midnight.mp3'
 316.mp3
'53rd & 3rd.mp3'
'747 (Strangers In The Night -- Live).mp3'
 80Times.mp3
'95 NASTY.mp3'
'99 Luft Balloons.mp3'
'99 Ways to Die.mp3'
'Ace of spades.mp3'
'Aces High.mp3'
'After All (The Dead).mp3'
'After Dark.mp3'

Wookie922, 2021-06-15:
Just got it to load needed the .. but not the ~

Wookie922, 2021-06-16:
So I figured out the music path issue.  I have two new problems that have consumed my afternoon.  one is that I have no sound I'm thinking that something is not configured in the pi correctly.  I'm hooked up to my amplifier via HDMI that appears to be correct I also installed mpg321 but I'm obviously missing something.  I tried pointing mpg321 at my Music file, maybe a mistake, it took the change but didn't create sound.  I tested the speakers using a sound test I don't remember at the moment and they worked. 
The second issue is that when i start up the skin the first song it plays(silently) continues indefinitely regardless of whats in the queue.  thanks  again mostly i just need a push in the correct direction

Witchboard, 2021-06-16:
Don't update the pi after you image it.  I ran into the same problem and re-imaged and didn't update the OS and it ran fine, so something in the update broke it.  That is presuming you updated it.

Wookie922, 2021-06-16:
I did

Witchboard, 2021-06-16:
Check out this thread and see if it helps.
https://github.com/chundermike/rpi-fruitbox/issues/61

Wookie922, 2021-06-16:
I will thank you.

Wookie922, 2021-06-18:
Thank you. Problem solved. I reformatted the as card. Burned raspian again. Then used its desktop to set up the pi. I then manually installed fruitbox following the directions here. Lastly I had to expand the available memory I think that is in the directions and I just skipped it for some reason. Everything is up and running now. I'm very excited.

closed 2021-06-21 by Wookie922