chundermike/rpi-fruitbox #87: No sound

j529, 2021-10-07:
RPi 3b+

I spent a lot of time getting things working a few months ago. Finally got around to building a cool custom case for it and have no audio now. I'm trying to play through headphone jack. Have tried with and without running raspi-config. Have tried running sudo and as pi. Have tried different skins. Tried adding a asound.conf file as one user suggested here. I do not have any kind of soundcard and am using the 7" touchscreen 

oxmplayer is what I used to test that the speaker is working through headphones, no problem. Not sure where else to turn on this one. 

Previously, I had some problems with running it from .bashrc, but if I raspi-config headphones and ran from command line, all was well. Now, I can't seem to get any audio to output at this point.

sergiohinniger, 2021-10-09:
If your RPi is updated maybe is an alsa issue. I had a similar problem that i fixed with the following solution (or workaround i dont know), if interest to anyone:

Here’s the step-by-step solution:

In the command line type:

```
sudo touch ~/.alsoftrc
sudo nano ~/.alsoftrc
```
Will open a file, paste this, save, exit:

```
[alsa]
mmap = false

```
then type:

```
sudo cp ~/.alsoftrc /etc/openal/alsoft.conf
sudo reboot
```

sergiohinniger, 2021-10-10:
Try then:

edit cmdline.txt

`sudo /boot/cmdline.txt
`
and add:

`snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_compat_alsa=1
`