chundermike/rpi-fruitbox #90: Issue running fruitbox

Mike12047, 2021-11-22:
Hello Mike,

Love fruitbox and ran into an issue and don't know how to proceed.  I just can't get it to run.
[image]

bmaselli13, 2021-11-22:
Getting the same error as well.

Mike12047, 2021-11-23:
I discovered that it was because I was using the latest os.  Once I
downloaded buster from the archives, and did the initialization of my
Raspberry Pi 2 I was able to install and run Fruit box with no issues!

bmaselli13, 2021-11-23:
Hmm. I tried with Buster and am still having shared library issues. Mind sharing your steps?

bmaselli13, 2021-11-23:
I reimaged with Raspian-buster-lite, booted up, went through the steps on the main page. When I try to run Fruitbox, I'm still getting the shared library error.  Running on a Rasp Pi b

bmaselli13, 2021-11-23:
Reinstalled each library and am now at a different error. Shows * failed to add service - already in use? I tried killall and it still shows up.

bmaselli13, 2021-11-24:
Resolved for me doing the following:

You need to enable the Legacy GL driver. You can do this by typing sudo raspi-config and then going into Advanced Options > GL Driver > Select Legacy.

Mike12047, 2021-11-24:
Sorry for the late reply.  I'm very new at all things Pi.  I installed
buster and it tells me it's the oldstable version so it won't update.  But
it runs fruitbox flawlessly so I'm happy with that.  I wish I could be more
help but not as versed in the programming as I want to be yet.  Hopefully
being oldstable doesn't mean it won't work down the road, but I guess we
will see how it goes.

ZS6TVB, 2021-11-24:
It will work fine down the road.
For a dedicated Jukebox you won't really need to do updates.

Powered by Mailbird [https://www.getmailbird.com/?utm_source=mbsig]

—
[654e3db3-48ef-43ab-aff1-e0d8d7a85d0a]

Mike12047, 2021-11-24:
I hadn't thought about it that way but you're absolutely right. I purchased
a gutted rock-ola wall box and I'm planning on making that into a fruit box
project so maybe I'll try it then with the latest OS

PICCHIO68, 2022-02-06:
I have the same issue, Advanced Options > GL Driver > Select
 Legacy "the GL driver is disabled" 
When I run Fruitbox.......same issue. Why? Thanks for help

Mike12047, 2022-02-06:
Not sure why but the latest release prohibits fruitbox from running. So I
used the legacy release and it works perfect.

Mike

PICCHIO68, 2022-02-06:
[image]
IT'is OK this ? Thanks I have rapberri pi 3 model B

Mike12047, 2022-02-06:
Hello again, Yes thats the one.  I am also running fruitbox on a 3b using
legacy release and I have had zero issues so far.  I hope it helps you, and
feel free to reach out if you need help.  I'm still fairly new at this, but
I will do what I can.

Mike

PICCHIO68, 2022-02-06:
Thank you very much. I try and I let you know. Ciao from Italy ! :)