chundermike/rpi-fruitbox #98: Database Sort Loses Album Artwork

richiep124, 2022-02-27:
I'm a newbie, working through a lot of issues with fruitbox, bit it's going ok so far.

I'm trying to use the Granite skin and have the correct song order in each album.

When I try to use tge .cfg file to Sort by album, and by tracknumber it just does not work.  It shows a single song on each album.

I've had success manually sorting some of the albums in the database, but I have 3000 songs in 250 albums...much too onerous to do manually.

I've also had success exporting the database, and sorting it with a db editor, but when I copy it back into fruitbox, none of my artwork displays.

The folder.jpg files are still there, but when I run fruitbox they just do not show up.

Any ideas?

richiep124, 2022-02-27:
One other question, is there a way to do a global database sort from with the raspberry, using nano or something and sort by album, by tracknumber??

chundermike, 2022-03-21:
Are your track numbers correct in your song mp3 tags? If they all are set to 1 then you'll only get a single track per album.

To display folder.jpg artwork, you must set the NowPlayingArtworkMode parameter in the skin cfg file to: Folder.