chundermike/rpi-fruitbox #99: Error about loading the shared libraries and libbrcmglesv2.so

Comet-boy, 2022-03-13:
New user here…. So most likely something basic that I'm doing wrong....

Just trying to setup Fruitbox using the recommendations on the Chundermike Github page that looks like this:
____________
#!/bin/bash
echo "----------------------"
echo "Installing fruitbox..."
echo "----------------------"
rm master*
rm -Rf rpi-fruitbox-master-backup
mv rpi-fruitbox-master rpi-fruitbox-master-backup
sudo apt-get -y install libsm-dev libxcursor-dev libxi-dev libxinerama-dev libxrandr-dev libxpm-dev libvorbis-dev libtheora-dev
wget https://codeload.github.com/chundermike/rpi-fruitbox/zip/master
unzip master
rm master
cd rpi-fruitbox-master
chmod +x ./fruitbox
echo
echo "Quick Start Guide"
echo "-----------------"
echo "1. Choose a skin (see skins directory); lets call it MYSKIN."
echo "2. Edit (using nano for example) the skins/MYSKIN/fruitbox.cfg file"
echo "   and change the MusicPath value to point to where your mp3 files are kept."
echo "3. run fruitbox..."
echo "   sudo ./fruitbox --cfg skins/MYSKIN/fruitbox.cfg"
echo ""
echo "For help, run fruitbox with no arguments (i.e. ./fruitbox)"
__________________

Everything seems OK and setups up according to the setup instructions on that page. I’ve changed/edited the Music path and edited the fruitbox skin I’m using to reflect that.

I when I attempt to run fruitbox from the command line (now from within the  rpi-fruitbox-master directory?)
using the command: "   sudo ./fruitbox --cfg skins/WallJuke/fruitbox.cfg"

I get an error about loading the shared libraries - libbrcmglesv2.so??

There are two things I notice different from the original older information on the Github page.....

1 – the OS version is the latest Debian Bullseye with desktop (2022-01-28)
2 - the new OS doesn’t have the (sudo raspi-config -> Advanced Options -> Memory Split) as mentioned? You can set the GUI size in the config setup but it has changed from the Advanced Options page….

Not sure where to go from here? Try with an older version of Pi maybe??

Any tips appreciated.

Comet-boy, 2022-03-13:
Just tried an older Pi version and I don't get that error about loading the shared libraries - libbrcmglesv2.so...... so must something to do with the current version?

chundermike, 2022-03-21:
try this from the command line,,,

cd /opt/vc/lib
sudo ln -s libbrcmEGL.so libEGL.so
sudo ln -s libbrcmGLESv2.so libGLESv2.so
sudo ln -s libbrcmOpenVG.so libOpenVG.so
sudo ln -s libbrcmWFC.so libWFC.so

Arcamat, 2022-04-08:
> try this from the command line,,,
> cd /opt/vc/lib
Hello
No such file or directory

Arcamat, 2022-04-08:
``

> try this from the command line,,,
> 
> ```
> cd /opt/vc/lib
> sudo ln -s libbrcmEGL.so libEGL.so
> sudo ln -s libbrcmGLESv2.so libGLESv2.so
> sudo ln -s libbrcmOpenVG.so libOpenVG.so
> sudo ln -s libbrcmWFC.so libWFC.so
> ```
Hello
No such file or directory

Comet-boy, 2022-05-09:
Back now on this project and trying to run FruitBox on a fresh Pi OS install - "Raspberry PI OS (32-bit) 2022-04-04" and Chyndermike this suggestion fails.... as mentioned by Arcamat - no directory found (cd /opt/vc/lib)???
Same error about libbrcmGLESv2.so when you try to run FruitBox. Old OS version does work but no Touch Screen support. I need to use a touch screen for my project....
Any ideas??????

chundermike, 2022-05-09:
Hi

Check out the build.txt in the build directory, but essentially...

cd /opt/vc/lib
sudo ln -s libbrcmEGL.so libEGL.so
sudo ln -s libbrcmGLESv2.so libGLESv2.so
sudo ln -s libbrcmOpenVG.so libOpenVG.so
sudo ln -s libbrcmWFC.so libWFC.so

If this doesn't work then it's down to the latest OS...the graphics driver has changed completely, and the Allegro library I'm using isn't keeping pace.  Probably worth reverting to the previous OS.  What touchscreen do you have? I'm aware that fruitbox doesn't work with ELO touchscreens but fruitbox v2 fixes this, which is out soon.

Mike

Comet-boy, 2022-05-10:
With the current version the /opt directory only has 2 subfolders - pigpio and WidevineCd-m
no vc/lib folder??

The screen is an ingcool 7IP-CAPLCD and runs fine in the pi windows enviroment.

Thanks for your help Mike! Greatly appreciated.

When is fruitbox v2 likely to be released? are there more Touch Screen options (skins) likely in it. TouchOne looks perfect for my job......

Comet-boy, 2022-05-10:
If anyone can suggest a version that does run correctly and supports Touch Screens, I happy to try that option as well.

Comet-boy, 2022-05-10:
I have now gone back to an earlier version of Raspian Buster (2019) and all runs fine with no errors.

But one question.... I can't seem to get the touch screen working. In setting up I follow the Userguide.txt..... the --config-buttons command line seems to work fine but the --calibrate-touch command seems to do nothing? The touch screen works fine outside Fruitbox. I think I'm missing something simple here! Is there more information on touch screen setup? I've tried TouchOne and various other iOtero skins. Screen is a 7" ingcool 7IP-CAPLCD.

Comet-boy, 2022-05-11:
That last question might be a stupid one!

I see you should run the two commands together on the command line 'sudo ./fruitbox --config-buttons --calibrate-touch'

Was a late night!!!