chundermike/rpi-fruitbox #105: Custom Skins

richiep124, 2022-04-17:
Mike, thanks for all your work on this.  I'm working on customizing a couple of the skins and had a question.

I understand in the config file how fruitbox knows what letters/numbers are going to be used to select songs.

But how does it know which key combos are linked with which songs?

For example, on the Wurly skin, how does it know that A1 is the first song on the screen, and A2 is the second one, etc. etc.??

chundermike, 2022-04-18:
Hi,

It depends on the order of the SelectButtons entries in the config file, and the order in which the [pages] are defined in the config file. If you add --debug-song-nums to the command line when running fruitbox (i.e. ./fruitbox --cfg <skin> --debug-song-nums), fruitbox will prefix each song name in the title strip with the select code, so you can see what they are.  Also, there is another config file option 'SelectButtonsSequence' which can be set to 'RowCol' or 'ColRow' which will swap the sequencing order.  Have a play with it and the --debug-song-nums command line option to see what effect it has.  Also note that you can define as many 'SelectButtons' strings as you like; the number of definitions determines the number of select digits.  So if you wanted a three-digit select code, you'd have three SelectButtons entries.  Also note that the string values for the SelectButtons can have missing letters or be in any order (ie. SelectButtons = ABCDEFGHJK, will omit the letter 'I', which some jukeboxes do to avoid confusing it with the number '1').  And you can just have numeric select codes with no letters (i.e. 00 to 99 is achieved by two SelectButtons = 0123456789 entries).

Hope this helps,
Mike

richiep124, 2022-04-18:
Got it.. thanks Mike.
Richie P.

Sent from the all new AOL app for Android