chundermike/rpi-fruitbox #106: Cannot load some skins

barrychilds, 2022-04-18:
Hi, I love fruitbox, but I am new to all this.  I have it all working but some skins 'iOtero, and both the Jays won't load. It says, Couldn't find configuration file to open. I really want a skin that will play CDs but only one track like SK jukebox. I am very new to this and find it all hard to understand. I hope you can help me. Barry

chundermike, 2022-04-18:
can you post the fruitbox.log file?

barrychilds, 2022-04-19:
[fruitbox.log](https://github.com/chundermike/rpi-fruitbox/files/8512487/fruitbox.log)
Thank you for your help. I hope I have posted the correct file. Barry

barrychilds, 2022-04-19:
Hi Mike. I see in the skins.txt the skins mentioned above are not in there. It seems there is no skin to play individual tracks from CDs. The granite skin only allows me to play the whole album.  Do you know of any skin that would do the job, and if so is it possible for me to download and install it?

chundermike, 2022-04-19:
Hi Barry,

I think due to a cock-up in my GitHub commit, the path to JaysCDJukebox is actually 'skins/JaysCDJukebox/JaysCDJukebox/fruitbox.cfg'.  Also fruitbox doesn't currently support the selection of individual tracks in Albums mode.  I hope to fix this in a future release.

barrychilds, 2022-04-20:
Thank you, Mike. I think Wurly will do the job anyway as it loads all the CDs as singles. There is just one more thing (and I promise to donate when my pension arrives at the end of the month), I have had no success on getting it to load at boot. I just don't understand the instructions from people, so I did try your all-in-one solution. Only trouble is I don't know how to get behind the skin to do anything. The keys only seem to operate the skin. Is there a desktop behind it? and how do I access it?
Thanks, Barry

barrychilds, 2022-04-20:
Thanks very much Mike. I just have a few simple issues that I hope you can resolve. I would like to add some info like keyboard layout (I live in France) internet settings etc. On the desktop version this all came up automatically for me to do. I also have a pimoroni onoff button. When I put in the line to make it work it says: Could not resolve host: get.pimoroni.com
Thank you. Barry

chundermike, 2022-04-20:
Hello Barry

Not 100% sure as not fruitbox related, but from the command line you could try typing:

sudo raspi-config

This will bring up a menu in which you can change all those types of settings.

Mike

barrychilds, 2022-04-20:
Thanks Mike for your quick reply. I will try tonight. I have to go out now. I have to travel 60 miles each day for cancer treatment. It disrupts everything. Barry

barrychilds, 2022-04-20:
Hi Mike, I somehow got everything done. My last problem is when I turn on, I get to the select skin every time. I want it to load Wurly on start up. If this can be done, please explain like you are talking to a five year old. Scrap that, two year old.
Barry

chundermike, 2022-04-20:
Hi Barry

The line that starts fruitbox will have the --choose-cfg skins.txt argument on it. Remove this and replace with --cfg skins/Wurly/fruitbox.cfg. I can't remember where the line is stored but it may be in ~/.bashrc (so from a command line type sudo nano ~/.bashrc to edit the file). Then Ctrl-x to save and quit.

If this doesn't work you may just be able to edit the skins.txt file and remove all skin entries apart from Wurly. I think if fruitbox sees just one entry in the skins.txt file it won't bother to run the skin chooser, but will just run up that skin immediately.

Mike

barrychilds, 2022-04-20:
I found a lot of lines of script and at the bottom in white was various things you can do, but I could not find the --choose-cfg skins.txt anywhere.

Barry

barrychilds, 2022-04-20:
Iv'e sent you £20 now Mike. I will send a further £20 each month for 5 months. So it will be £100 in total. I hope this helps.
Barry

barrychilds, 2022-04-20:
Actually, Mike. I think i'll leave it as it is. It uses the same buttons I will be configuring. So it's very good to be able to swap skins.
I have an old Wurlitzer CD jukebox that I have ripped the inners out of. (except the amp and speakers). The idea is to put everything inside the jukebox, with a monitor instead of the original cards. I hope to use the jukebox buttons plus a few extras.
I will try to configure the buttons and will ask for help if I find it too daunting. Thank you for all your help.
Barry

chundermike, 2022-10-11:
Wow, thank you Barry, that's extremely generous. Much appreciated.  I am due to release version 2 soon, which should make things easier for you. But please, continue to ask for help. When I get the chance soon I will double check on the information about getting the Wurly skin to boot from power up. Also, because of your generosity, I would be most happy to try and implement any new features you would like, assuming they are possible.

Many thanks again
Mike