chundermike/rpi-fruitbox #108: Autorun of command line?

Comet-boy, 2022-05-22:
[image]
Hi Mike,  just finished a the “Fruitbox” project plus the 3D printed case and all the hardware etc.
The problem is that I can’t seem to get the command line below to autorun on start -up.
Sudo ./fruitbox –cfg skins/TouchOne/fruitbox.cfg
Just using raspbian-buster….. pi 3b+ with a 7” touch screen. Fruitbox runs 100% as per your guide.
What is the recommended way to do this??
I think I’m missing something obvious.
Also I see Version 2 is available for download. Are there any changes to the Touch screen interface??

chundermike, 2022-05-24:
Hi,

Looks amazing! Nice one. When you say you can't get it to run, what exactly is the issue; is it giving an error or just not doing anything? Where have you put the command line?  The last section of the userguide.txt file should explain how to do it.

v2 executable is almost ready, the changes to the touchscreen interface are...

1. ELO touchscreens are now supported.
2. Calibration is automatically run on first time use. 
3. Button touch animations are simpler and don't require skin status objects.
4. Buttons no longer need to specifically mapped as "Touch"; they will be automatically mapped to touch if a touchscreen has been detected and their touch areas are defined in the skin file.

Basically the touch changes will make it much easier for the user to manage touch screens and touch buttons, and design touch controls into their skins.

Cheers,
Mike

Comet-boy, 2022-05-24:
Thanks for the reply....

Just got it working:

Used nano to edit ( sudo crontab -e)
I add the line: @reboot (cd /home/pi/rpi-fruitbox-master; ./fruitbox –cfg skins/TouchOne/fruitbox.cfg ) /tmp/fruitbox.log 2>&1

Seems fine… but maybe crontab isn’t the best way to go.

Cheers
Bruce

PS look forward to playing with V2, are there an new touch skins??